pymodbus-dev/pymodbus #1786: Missing loop parameter for Modbus servers

h3ndrk, 2023-09-21:
### Versions

- Python: 3.10.12
- OS: Ubuntu 22.04
- Pymodbus: 3.5.2

### Pymodbus Specific

- Server: tcp and rtu - sync and async

### Description

I'm missing a way to inject an already created asyncio loop into the Modbus server. It seems to be removed by https://github.com/pymodbus-dev/pymodbus/pull/1714. The server now always uses the current loop which might not be available if the server is instantiated outside of the event loop and started later. Problematic code:

https://github.com/pymodbus-dev/pymodbus/blob/dd2bfdef1b3d289d75a6709ea66af3c9fb6f9322/pymodbus/server/async_io.py#L275

This is the code that we are trying to get working:

```python
from asyncio import new_event_loop, run_coroutine_threadsafe
from threading import Event, Thread
from types import TracebackType

from pymodbus.datastore.context import ModbusServerContext
from pymodbus.framer import ModbusRtuFramer
from pymodbus.server.async_io import ModbusSerialServer

class ModbusInterface:
    def __init__(self, context: ModbusServerContext) -> None:
        self._loop = new_event_loop()
        self._server = ModbusSerialServer(context, ModbusRtuFramer, port="/dev/ttyUSB0", baudrate=9600)
        self._server_started = Event()

    def __enter__(self) -> "ModbusInterface":
        """Starts the PyModbus server and listens for incoming requests."""

        self._thread = Thread(target=self._loop.run_until_complete, args=(self._serve_forever(),))
        self._thread.start()
        self._server_started.wait()
        return self

    def __exit__(
        self,
        _exception_type: type[BaseException] | None,
        _exception: BaseException | None,
        _traceback: TracebackType | None,
    ) -> None:
        """Stops the PyModbus server."""

        run_coroutine_threadsafe(self._server.shutdown(), self._loop)
        self._server_started.clear()
        self._thread.join()

    async def _serve_forever(self) -> None:
        await self._server.transport_listen()
        self._server_started.set()
        await self._server.serving
```

We need to execute the server in a background thread started by the context manager. The problem is that the constructor of the `ModbusSerialServer` calls `asyncio.get_current_loop()` which is not available in the synchronous constructor.

What would probably solve the issue is to add a parameter again to the constructor of the server which takes an existing loop and only if this is not passed, get the current loop via `async.get_current_loop()`.

If you're happy with this solution, I could provide a PR if you want.

janiversen, 2023-09-21:
We call get_running_loop on purpose, that should work across threads (at least it does in my setup). We do not want a loop parameter. I am not too sure why you think it would solve your problem, because if there is a asyncio loop running, asyncio.get_running_loop() will return it.

The reason we do not want a loop parameter is exactly due to the fact that a running loop is needed, and we have had a lot problems due to the loop parameter pointing at loops not asyncio or not running.

a simple solutions is to secure that the async loop is started before you call ModbusSerialServer (might be in another thread), and since you suggest it as a parameter it must be running.

You can also have your own serve_forever, have a AsyncModbusSerialServer call there and then call _loop.server_forever()

janiversen, 2023-09-21:
I hope you are aware that mixing threads an asyncio is not recommended on the contrary. If you have an async environment use create_task instead.

h3ndrk, 2023-09-22:
Thanks for your answers!

Interesting, I think we also tried to abuse to pass a non-running loop to the server.

We have the following requirements to our `ModbusInterface`:

1. We need to inject the server (RTU/TCP and connection parameters) into it (cannot be seen in my simple example)
2. We don't have a running loop available at dependency injection
3. We want to manage the resources of background threads and asyncio loops in the context manager (later than the constructor)
4. We need to wait (in `__enter__()`) until the server has started properly (therefore the custom `serve_forever()`)

Since we want to construct the server ahead of time (synchronous dependency injection), we tried constructing it with a non-running loop and then move this loop into a thread to run it there.

It seems we need to figure out a way to delay the construction of the server until our whole background thread and asynchronous environment has spun up (weaken requirement 2).

> I hope you are aware that mixing threads an asyncio is not recommended on the contrary. If you have an async environment use create_task instead.

Yes, in general I agree. In our case, we have a synchronous environment and need to execute your server and need a custom `serve_forever()` which requires us to write a coroutine and think about proper sync/async interfacing.

> The reason we do not want a loop parameter is exactly due to the fact that a running loop is needed

Fulfilling your requirement would still enable a loop parameter if you would check/assert in the constructor [if the loop is actually running](https://docs.python.org/3/library/asyncio-eventloop.html#asyncio.loop.is_running). That would not solve our underlying problem though.

> We call get_running_loop on purpose, that should work across threads (at least it does in my setup).

[`asyncio.get_running_loop()`](https://docs.python.org/3/library/asyncio-eventloop.html#asyncio.get_running_loop) will "return the running event loop in the current OS thread". So that will probably *not* work across threads.

My current approach to all of this would be to introduce some connection parameter class which we can pass to our `ModbusInterface` and it would create the server later in a coroutine. That breaks dependency injection but there is no other way it seems.

janiversen, 2023-09-23:
Take a look at startSerialServer that shows a way around some of the problem.

h3ndrk, 2023-09-27:
Thanks for the hint but that does not solve all of our requirements. Nevertheless we have implemented a solution now (as described above), therefore closing. Thanks again!

closed 2023-09-27 by h3ndrk