pymodbus-dev/pymodbus #1790: Client and Server on same Serial device/transport

ashb, 2023-10-01:
### Versions

- Python: 3.11
- OS: linux
- Pymodbus: 3.5.2
- Modbus Hardware (if used): Somewhat generic USB serial device off Amazon

### Pymodbus Specific

- Server: rtu -async
- Client: rtu- async

### Description

I am trying to write both a server and a client in the same process that communicate over the same (and only) USB serial device.

The goal is to be a modbus Server and respond to my SolarEdge inverter and to "translate" the registers from my Eastron SDM120 meter.

This project(s) exist already

- https://github.com/nmakel/sdm_modbus/
- https://github.com/nmakel/solaredge_meterproxy/tree/master

However if I configure the semp-rtu to use the same device path it doesn't work, as mentioned in https://github.com/pymodbus-dev/pymodbus/issues/1606 (and as expected) this doesn't work.

The desired behaviour I would like is:

When a read register request comes in on slave_id 2 (for example) -  we make a read register request to slave_id 247, take the returned values and translate/transform them into the addresses that SolarEdge expects.

If doing it in response to a request isn't possible, a fallback would be to read the registers from 247 every n seconds and have the server response to 2 just using the latest values.

I tried a somewhat naive approach of creating a server and just sending stuff to the `self.transport_send` but either that doesn't send anything, or the responses get dropped.

Do I perhaps need to go lower level and use the Transport interface directly?

janiversen, 2023-10-02:
You cannot have 2 program parts sharing the same serial device. Apart from that the modbus protocol only allows 1 client (controlling the communication) and many servers.

janiversen, 2023-10-02:
You basically wants to have 2 clients on the same line, which is breaking the protocol, and also how rs485 works.

ashb, 2023-10-02:
I've misunderstood modbus then --  I thought it was possible to have multiple masters on a bus as long as they don't speak to the same device?

(Thanks for the response!)

janiversen, 2023-10-02:
No it's in the nature of both RS485 and the modbus protocol.

janiversen, 2023-10-02:
Not supported by modbus.

closed 2023-10-02 by janiversen