pymodbus-dev/pymodbus #1792: read_holding_registers can't be awaited, but returns coroutine

doxg, 2023-10-06:
- Python: 3.9
- OS: Windows 10
- Pymodbus: 3.1.3

### Pymodbus Specific

- Server: Async Serial
- Client: Async Serial

### Description

Greetings! I have previously used pymodbusTCP lib, but it's my first time using pymodbus.

I am trying to implement simple server-client connection simulation, where client writes values to registers and then reads. So far, there were no problems writing value to registers, but I am stuck on reading values. The _read_holding_registers()_ method of **AsyncModbusSerialClient** class is defined as regular method, however its return value is coroutine. When I try to await result (I've seen in examples https://github.com/pymodbus-dev/pymodbus/blob/dev/examples/client_async.py, also in some issues), my IDE-Pycharm shows hint:
"Class 'ReadHoldingRegistersResponse' does not define '__await__', so the 'await' operator cannot be used on its instances ".
So when I run code using try-catch block, the Exception is caught as Generic Exception, but I can't access its message (Exception is considered as None). However if I don't await r_ead_holding_registers(),_ it returns coroutine and I need to access value. I tried to wrap it in task and use asyncio.ensure_future() which also didn't work. 

May be I'm missing some simple stuff. 
Any help would be appreciated.

### Code and Logs

**server.py**

```
import asyncio
from pyModbusTCP.utils import *
from pymodbus.datastore import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusServerContext, ModbusSlaveContext
from pymodbus.server import StartAsyncSerialServer

from utils import read_args, convert_unicode

async def run_server():
    server2 = await StartAsyncSerialServer(port="COM3", method="rtu", baudrate=9600, stop_bits=1, bytesize=8)
    server2.serve_forever()

asyncio.run(run_server())
```

**serial_client.py**

```
import asyncio
import random
import pymodbus
from pyModbusTCP.utils import encode_ieee, decode_ieee, \
    long_list_to_word, word_list_to_long
from pymodbus.client import AsyncModbusSerialClient
from pymodbus.exceptions import ModbusException
from utils import read_args

pymodbus.pymodbus_apply_logging_config()

class FloatModbusClient:
    """A ModbusClient class with float support."""

    def __init__(self, **kwargs):
        self.client = AsyncModbusSerialClient(**kwargs)

    async def read_float(self, address, number=1):
        """Read float(s) with read holding registers."""
        try:
            reg_l = await self.client.read_holding_registers(address, number * 2, slave=0)

            if reg_l.registers:
                return [decode_ieee(f) for f in word_list_to_long(reg_l.registers)]
            else:
                return None

        except ModbusException as modbus_exception:
            print("Modbus Exception:", modbus_exception)
        except Exception as excep:
            print("Generic Exception:", excep)

    async def write_float(self, address, floats_list):
        """Write float(s) with write multiple registers."""
        b32_l = [encode_ieee(f) for f in floats_list]
        b16_l = long_list_to_word(b32_l)
        self.client.write_registers(address, b16_l, slave=0)

    async def write_pack(self, args):
        rand_num = random.randint(1, 100)
        for category, address in args.reg_addr.items():
            await self.write_float(address, [rand_num])
            print(f'Successfully wrote {rand_num} to {address}')

async def run_helper():
    args = read_args('./configs/configs.yaml')
    clientWrapper = FloatModbusClient(port="COM1", baudrate=9600, stopbits=1)
    await clientWrapper.client.connect()
    if clientWrapper.client.connected:
        print("Client is connected")
    try:
        while True:
            await clientWrapper.write_pack(args)
            await asyncio.sleep(1)
            float_l = await clientWrapper.read_float(200, 1)
            print("Printing values: ", float_l)
            await asyncio.sleep(1)
    except Exception as e:
        print(e)
    finally:
        await clientWrapper.client.close()

if __name__ == "__main__":
    asyncio.run(run_helper())
```
    
    
The console output is: 
Generic Exception 
Printing values:  None
.... (and so on)

janiversen, 2023-10-06:
Please have a look at our examples, you at least miss a .connect()

When using the async classes, request can and should be awaited.

janiversen, 2023-10-06:
you should actually also have an "await" when calling "self.client.write..."

doxg, 2023-10-06:
> Please have a look at our examples, you at least miss a .connect()
> 
> When using the async classes, request can and should be awaited.
> 
> the example "simple_async_client.py" should do what you want.
> 
> I have no idea how the lib you used before works, that is another project.

Thank you for your reply.
Yeah, I took a look at examples you provided. That's why I am confused.
The client connection is done after I declare class instance:
```
await clientWrapper.client.connect()
if clientWrapper.client.connected:
    print("Client is connected")
```

and when I add await to "self.client.write_registers(address, b16_l, slave=0)" it shows the same warning/hint:
Class 'WriteMultipleRegistersResponse' does not define '__await__', so the 'await' operator cannot be used on its instances)

janiversen, 2023-10-06:
What does the general exception say exactly, it's hard to help with only partial information.

doxg, 2023-10-06:
> Versión 3.1.3 is a pretty old release, current release is 3.5.2 so first step should be to upgrade, because we have made many changes to the async classes.
> 
> The calls are not declared async (a method do not define await), because they are shared between the sync and the async version. However the async version always return a future as you have noted. That is perfectly legal python code and the "sharing code trix" is used in many places in the standard libs.
> 
> Your code looks ok. You include pymodbusTCP which is not really needed, pymodbus offers similar functions.

Thank you for your fast reply. 
I just upgraded pymodbus lib to 3.5.2 and got log statements. Also I added to "await self.client.write_registers(address, b16_l, slave=0)". The log statements that I've got:

```
2023-10-06 16:48:01,732 DEBUG logging:102 Connecting to COM1.
2023-10-06 16:48:01,732 DEBUG logging:102 Connecting comm
2023-10-06 16:48:01,732 DEBUG logging:102 Connected to comm
2023-10-06 16:48:01,732 DEBUG logging:102 callback_connected called
2023-10-06 16:48:01,732 DEBUG logging:102 send: 0x0 0x10 0x0 0xc8 0x0 0x2 0x4 0x42 0x9c 0x0 0x0 0x2e 0xf3
2023-10-06 16:48:01,732 DEBUG logging:102 Adding transaction 0
Client is connected
2023-10-06 16:48:04,723 DEBUG logging:102 send: 0x0 0x10 0x0 0xc8 0x0 0x2 0x4 0x42 0x9c 0x0 0x0 0x2e 0xf3
2023-10-06 16:48:04,723 DEBUG logging:102 Adding transaction 0
2023-10-06 16:48:07,721 DEBUG logging:102 send: 0x0 0x10 0x0 0xc8 0x0 0x2 0x4 0x42 0x9c 0x0 0x0 0x2e 0xf3
2023-10-06 16:48:07,721 DEBUG logging:102 Adding transaction 0
2023-10-06 16:48:10,729 DEBUG logging:102 send: 0x0 0x10 0x0 0xc8 0x0 0x2 0x4 0x42 0x9c 0x0 0x0 0x2e 0xf3
2023-10-06 16:48:10,729 DEBUG logging:102 Adding transaction 0
2023-10-06 16:48:13,728 DEBUG logging:102 Connection lost comm due to Server not responding
2023-10-06 16:48:13,728 DEBUG logging:102 Getting transaction 0
Modbus Error: [Input/Output] ERROR: No response received after 3 retries

Process finished with exit code 0
```
PS: I am using Virtual Serial Port Driver software, and I can monitor that COM3 and COM1 are being paired by the server and serial_client scripts.

janiversen, 2023-10-06:
Your server calls serve_forever() why ? that is not as in the examples and wrong,

You probably need to define a datastore with the addresses/slaves you want to use, or secure the you use slave and addresses in our default

slave=0 is not correct with the default it should be slave=1 again see our see our examples,

janiversen, 2023-10-06:
Did you look at the server examples and our documentation it seems not, framer="rtu" is not supported.

doxg, 2023-10-06:
> Did you look at the server examples and our documentation it seems not, framer="rtu" is not supported.

Yeah, I skipped the part for datablock configurations and context, which was foolish. I will study them one more time. 
Thank you for your time and efforts.

PS: Can you leave this issue open? In case, if I don't figure out solution.

janiversen, 2023-10-06:
Issues are for reporting bugs, in this case the problem seems to be in op your code so closing.

Discussions are more general.

closed 2023-10-06 by janiversen