pymodbus-dev/pymodbus #1794: Setting reconnect_delay=0 in AsyncModbusTcpClient constructor causes infinite busy-loops on connection loss

sthelen-enqs, 2023-10-06:
### Versions

- Python: 3.11.4
- OS: Ubuntu 20.04
- Pymodbus: 3.5.2
- Modbus Hardware (if used): N/A

### Pymodbus Specific

- Server:  tcp - async
- Client: tcp - async

### Description

I tried setting reconnect_delay=0 in the constructor for AsyncModbusTcpClient as documented in https://pymodbus.readthedocs.io/en/latest/source/client.html#pymodbus.client.ModbusBaseClient to prevent the client from reconnecting if the TCP connection drops, but whenever the connection dropped pymodbus would instead busy-loop at 100%CPU forever.

### Code and Logs

Minimal Example:
```python
import asyncio

import pymodbus
from pymodbus.client import AsyncModbusTcpClient

pymodbus.pymodbus_apply_logging_config()

async def main():
    client = AsyncModbusTcpClient(host="127.0.0.1", port=5502, reconnect_delay=0)
    r = await client.connect()

    while True:
        if not client.reconnect_task:
            resp = await client.read_input_registers(13, 2, slave=1)
        await asyncio.sleep(5)

if __name__ == "__main__":
    asyncio.run(main())

```

Logs (repeat ad-infimum):
```
2023-10-06 17:00:22,639 DEBUG logging:102 Connecting to 127.0.0.1:5502.
2023-10-06 17:00:22,639 DEBUG logging:102 Connecting comm
2023-10-06 17:00:22,640 DEBUG logging:102 Connected to comm
2023-10-06 17:00:22,640 DEBUG logging:102 callback_connected called
2023-10-06 17:00:22,640 DEBUG logging:102 send: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0xd 0x0 0x2
2023-10-06 17:00:22,640 DEBUG logging:102 Adding transaction 1
2023-10-06 17:00:22,640 DEBUG logging:102 recv: 0x0 0x1 0x0 0x0 0x0 0x7 0x1 0x4 0x4 0x0 0xe 0x0 0xf old_data:  addr=None
2023-10-06 17:00:22,640 DEBUG logging:102 Processing: 0x0 0x1 0x0 0x0 0x0 0x7 0x1 0x4 0x4 0x0 0xe 0x0 0xf
2023-10-06 17:00:22,641 DEBUG logging:102 Factory Response[ReadInputRegistersResponse': 4]
2023-10-06 17:00:22,641 DEBUG logging:102 Getting transaction 1
2023-10-06 17:00:27,646 DEBUG logging:102 send: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0xd 0x0 0x2
2023-10-06 17:00:27,647 DEBUG logging:102 Adding transaction 2
2023-10-06 17:00:27,647 DEBUG logging:102 recv: 0x0 0x2 0x0 0x0 0x0 0x7 0x1 0x4 0x4 0x0 0xe 0x0 0xf old_data:  addr=None
2023-10-06 17:00:27,647 DEBUG logging:102 Processing: 0x0 0x2 0x0 0x0 0x0 0x7 0x1 0x4 0x4 0x0 0xe 0x0 0xf
2023-10-06 17:00:27,648 DEBUG logging:102 Factory Response[ReadInputRegistersResponse': 4]
2023-10-06 17:00:27,648 DEBUG logging:102 Getting transaction 2
2023-10-06 17:00:28,814 DEBUG logging:102 -> transport: received eof
2023-10-06 17:00:28,815 DEBUG logging:102 Connection lost comm due to None
2023-10-06 17:00:28,815 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,815 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,816 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,816 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,816 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,817 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,817 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,817 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,818 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,818 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,818 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,819 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,819 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,819 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,820 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,820 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,820 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,821 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,821 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,821 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,822 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,823 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,823 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,823 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,823 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,824 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,824 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,824 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,824 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,825 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,825 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,825 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,826 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,826 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,826 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,826 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,826 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,826 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,827 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,827 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,827 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,827 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,827 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,827 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,828 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,828 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,828 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,828 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,828 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,828 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,828 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,828 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,828 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,829 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,829 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,829 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,829 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,829 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,829 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,830 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,830 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,830 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,830 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,830 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,830 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,830 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
2023-10-06 17:00:28,830 DEBUG logging:102 Wait comm 0 ms before reconnecting.
2023-10-06 17:00:28,831 DEBUG logging:102 Connecting comm
2023-10-06 17:00:28,831 WARNING logging:108 Failed to connect [Errno 111] Connect call failed ('127.0.0.1', 5502)
```

For the server I used the following code and then just killed it with ^C after the first few successful reads:
```python
import asyncio

from pymodbus.datastore import (
    ModbusSequentialDataBlock,
    ModbusServerContext,
    ModbusSlaveContext,
)
from pymodbus.server import StartAsyncTcpServer

datablock = ModbusSequentialDataBlock(0, list(range(25)))

ctx = ModbusServerContext(
    slaves=ModbusSlaveContext(di=datablock, co=datablock, hr=datablock, ir=datablock),
    single=True,
)

asyncio.run(
    StartAsyncTcpServer(
        context=ctx, address=("127.0.0.1", 5502), ignore_missing_slaves=True
    )
)
```

janiversen, 2023-10-06:
Sounds like you have found a bug, thanks for reporting it.

I have enough information to debug and hopefully find/repair the problem.

janiversen, 2023-10-06:
Please give your program a go, with this patch installed:

https://github.com/pymodbus-dev/pymodbus/pull/1795

It should work, but I do not have time to test it thoroughly right now.

janiversen, 2023-10-06:
Sorry I was too fast, this broke something. I will make a new version tomorrow.

janiversen, 2023-10-06:
If you want to test use #1795 

It will be part of either 3.5.3 or 3.6.0, but there are no confirmed time at this moment.

closed 2023-10-06 by janiversen

sthelen-enqs, 2023-10-09:
Thanks for the quick fix, seems to look good so far in tests.

PR #1795 (2023-10-06): Reconect_delay == 0, do not reconnect.

fixes #1794