pymodbus-dev/pymodbus #1806: Fault with config

cyberpater84, 2023-10-09:
### Versions

- Python: 3.11
- OS: 2023.10.1
- Pymodbus: 3.5.2
- Modbus Hardware (if used): Ecoforest

### Pymodbus Specific

- Server: tcp/rtu/ascii - sync/async
- Client: tcp/rtu/ascii - sync/async

### Description

When I use this integration I get a fault with reboot of the configuration YAML

### Code and Logs

`Logger: homeassistant.components.websocket_api.http.connection
Source: components/websocket_api/commands.py:226
Integration: Home Assistant WebSocket API ([documentation](https://www.home-assistant.io/integrations/websocket_api), [issues](https://github.com/home-assistant/core/issues?q=is%3Aissue+is%3Aopen+label%3A%22integration%3A+websocket_api%22))
First occurred: 08:33:41 (3 occurrences)
Last logged: 15:24:21

[139874146179904] The system cannot restart because the configuration is not valid: Component error: modbus - cannot import name 'ModbusBaseClient' from 'pymodbus.client' (/usr/local/lib/python3.11/site-packages/pymodbus/client/__init__.py)
[139874142896960] The system cannot restart because the configuration is not valid: Component error: modbus - cannot import name 'ModbusBaseClient' from 'pymodbus.client' (/usr/local/lib/python3.11/site-packages/pymodbus/client/__init__.py)
[139874139856704] The system cannot restart because the configuration is not valid: Component error: modbus - cannot import name 'ModbusBaseClient' from 'pymodbus.client' (/usr/local/lib/python3.11/site-packages/pymodbus/client/__init__.py)
Traceback (most recent call last):
  File "/usr/src/homeassistant/homeassistant/components/websocket_api/commands.py", line 226, in handle_call_service
    await hass.services.async_call(
  File "/usr/src/homeassistant/homeassistant/core.py", line 2012, in async_call
    response_data = await coro
                    ^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/core.py", line 2049, in _execute_service
    return await target(service_call)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/src/homeassistant/homeassistant/components/hassio/__init__.py", line 599, in async_handle_core_service
    raise HomeAssistantError(
homeassistant.exceptions.HomeAssistantError: The system cannot restart because the configuration is not valid: Component error: modbus - cannot import name 'ModbusBaseClient' from 'pymodbus.client' (/usr/local/lib/python3.11/site-packages/pymodbus/client/__init__.py)`

`
modbus:
  - name: "ecoforest"
    type: rtuovertcp
    host: IP_ADDRESS
    port: 502
    binary_sensors:
    - name: heatpump_alarm
      slave: 17
      address: 50
    - name: Heatpump Summer
      slave: 17
      address: 127
    - name: Heatpump Winter
      slave: 17
      address: 128
    sensors:
    - name: Heatpump Buitentemperatuur
      unit_of_measurement: °C
      scale: 0.1
      precision: 1
      slave: 17
      address: 11
      data_type: int16
    - name: Heatpump Boiler
      unit_of_measurement: °C
      scale: 0.1
      precision: 1
      slave: 17
      address: 8
[... 359 lines cut ...]
    - name: Heatpump Electrical Consumption october
      unit_of_measurement: kWh
      slave: 17
      address: 5177
      data_type: int16
    - name: Heatpump Electrical Consumption november
      unit_of_measurement: kWh
      slave: 17
      address: 5178
      data_type: int16
    - name: Heatpump Electrical Consumption december
      unit_of_measurement: kWh
      slave: 17
      address: 5179
      data_type: int16

    - name: Heatpump Number Of Starts L
      slave: 17
      address: 5281
      data_type: int16
    - name: Heatpump Number Of Starts H
      slave: 17
      address: 5282
      data_type: int16
`

janiversen, 2023-10-09:
This is the pymodbus project, not the homeassistant project ! pymodbus does not use configuration.yaml

But from personal experience I can tell you that you have a faulty homeassistant installation, seems it never installed the modbus component.

OR you have a custom component that conflicts with the modbus component.

cyberpater84, 2023-10-09:
You are right there is a modbus component that conflicting, it is this one https://github.com/wimb0/home-assistant-saj-modbus
After a uninstall of this applications the main modbus is working properly

closed 2023-10-09 by cyberpater84