pymodbus-dev/pymodbus #1821: ModbusClientMixin type annotations incompatibles with the async client

jvelo, 2023-10-16:
### Versions

- Python: 3.12
- OS: macos
- Pymodbus: 3.5.4
- Mypy : 1.5.1

### Pymodbus Specific

- Client: async

### Description

Mypy fails to validate standard async client calls due to incompatible type annotations in ModbusClientMixin.

### Code and Logs

Following code

```python
client = AsyncModbusTcpClient(HOST)
await client.connect()
coils = await client.read_coils(address=0, count=1, slave=SLAVE_ID)
```

Makes mypy fails with:

```
__main__.py:18: error: Incompatible types in "await" (actual type "ModbusResponse", expected type "Awaitable[Any]")
```

janiversen, 2023-10-16:
that because we use the same method for both sync and async, which is actually very normal.

We have configured mypy to the checks we want, you might want to do something similar.

closed 2023-10-16 by janiversen

janiversen, 2023-10-16:
If I may suggest, make a pull request where you have just changed a couple....just to secure that CI accept your changes, and then add a commit with all calls.

jvelo, 2023-10-16:
Sounds good, I'll have a try at it

jvelo, 2023-10-19:
After a little back and force, looks like it works for `mypy`, but still there's pytest failing for one version of python (3.10) – not sure what to make of it and if it's caused by the change itself : https://github.com/pymodbus-dev/pymodbus/actions/runs/6576303048/job/17865375534