pymodbus-dev/pymodbus #1851: No Response received from the remote slave

joshua-gallagher, 2023-10-24:
Pymodbus 3.5.4 fails to decode data coming from one of my ModbusTCP servers. I believe this is a bug in pymodbus, and not in the modbus server, because I am able to successfully retrieve the data using two other utilities:
- https://github.com/epsilonrt/mbpoll
- https://www.modbustools.com/modbus_poll.html
### Versions
- Python: 3.10.4
- OS: Windows 10, version 1809, OS build 17763.4974
- Pymodbus: 3.5.4

### Pymodbus Specific
- Client: Synchronous TCP

I have attached three network captures:
- pymodbus-working.pcapng: An example of pymodbus working against another ModbusTCP server I have. (Note that the modbus TCP port is 1030.)
- pymodbus-not-working.pcapng: This shows the problem I'm experiencing. Using pymodbus client, a single register is requested. A response comes back from the server with the data. 3 seconds later, pymodbus sends a FIN to tear down the session. (Note that the modbus TCP port is 9001.)
- mbpoll.pcapng: A network capture taken when I used mbpoll to verify that the data comes through in that utility okay.

[captures.tar.gz](https://github.com/pymodbus-dev/pymodbus/files/13127915/captures.tar.gz)

I ran the following Python code (with just a change to the IP and port address) to receive my one register successfully in the first case, and unsuccessfully in the second case:

```import pymodbus
from pymodbus.client import ModbusTcpClient as ModbusSyncClient
pymodbus.pymodbus_apply_logging_config()

client = ModbusSyncClient('192.168.1.100', port=9001)
client.connect()
rr = client.read_holding_registers(0, 1, slave=1)
print(rr)
```

The output in the failure case was:
```2023-10-24 18:36:23,294 DEBUG logging:102 Connection to Modbus server established. Socket ('192.168.1.199', 55940)
2023-10-24 18:36:23,294 DEBUG logging:102 Current transaction state - IDLE
2023-10-24 18:36:23,295 DEBUG logging:102 Running transaction 1
2023-10-24 18:36:23,297 DEBUG logging:102 SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0x1
2023-10-24 18:36:23,297 DEBUG logging:102 New Transaction state "SENDING"
2023-10-24 18:36:23,298 DEBUG logging:102 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2023-10-24 18:36:26,350 DEBUG logging:102 Incomplete message received, Expected 17 bytes Received 11 bytes !!!!
2023-10-24 18:36:26,351 DEBUG logging:102 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2023-10-24 18:36:26,354 DEBUG logging:102 RECV: 0x0 0x1 0x0 0x0 0x0 0xb 0x1 0x3 0x2 0x2c 0x4d
2023-10-24 18:36:26,357 DEBUG logging:102 Processing: 0x0 0x1 0x0 0x0 0x0 0xb 0x1 0x3 0x2 0x2c 0x4d
2023-10-24 18:36:26,359 DEBUG logging:102 Frame check failed, ignoring!!
2023-10-24 18:36:26,360 DEBUG logging:102 Resetting frame - Current Frame in buffer - 0x0 0x1 0x0 0x0 0x0 0xb 0x1 0x3 0x2 0x2c 0x4d
2023-10-24 18:36:26,363 DEBUG logging:102 Getting transaction 1
2023-10-24 18:36:26,365 DEBUG logging:102 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response
```

For comparison, the output I got from [mbpoll](https://github.com/epsilonrt/mbpoll) showed a successful read of the register:
```C:\Program Files (x86)\MBPoll>mbpoll 192.168.1.100 -a 1 -p 9001 -r 1 -c 1 -1 -t 4
mbpoll 1.5-2 - ModBus(R) Master Simulator
Copyright (c) 2015-2023 Pascal JEAN, https://github.com/epsilonrt/mbpoll
This program comes with ABSOLUTELY NO WARRANTY.
This is free software, and you are welcome to redistribute it
under certain conditions; type 'mbpoll -w' for details.

Protocol configuration: ModBus TCP
Slave configuration...: address = [1]
                        start reference = 1, count = 1
Communication.........: 192.168.1.100, port 9001, t/o 1.00 s, poll rate 1000 ms
Data type.............: 16-bit register, output (holding) register table

-- Polling slave 1...
[1]:    11261
```

I notice in the working case that the "Read Holding Registers" request receives an ACK from the server first, then the server sends the data in a subsequent frame with the PSH/ACK flags set. In the failure case, the server responds with the data as well as the PSH/ACK flags set in the same frame.

janiversen, 2023-10-25:
The problem seems to be this:

"Incomplete message received, Expected 17 bytes Received 11 bytes !!!!"

could it be that you have a non standard modbus device that delivers more than 2 bytes pr register?

janiversen, 2023-10-25:
I will take a closer look later but please supply a more complete log, this one is either edited or cut....because I miss a couple of important log entries.

janiversen, 2023-10-25:
I cannot see how/when you receive 17bytes, which is the real problem.  I will have a look at the capture files later

joshua-gallagher, 2023-10-25:
I just re-ran the test, this time requesting 10 registers. What's interesting is that the response, once again, is off by 6 bytes. Capture attached, and log file below:
```
2023-10-25 14:21:10,899 DEBUG logging:102 Connection to Modbus server established. Socket ('192.168.1.199', 56405)
2023-10-25 14:21:10,900 DEBUG logging:102 Current transaction state - IDLE
2023-10-25 14:21:10,901 DEBUG logging:102 Running transaction 1
2023-10-25 14:21:10,902 DEBUG logging:102 SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x3 0x0 0x0 0x0 0xa
2023-10-25 14:21:10,903 DEBUG logging:102 New Transaction state "SENDING"
2023-10-25 14:21:10,903 DEBUG logging:102 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2023-10-25 14:21:13,905 DEBUG logging:102 Incomplete message received, Expected 35 bytes Received 29 bytes !!!!
2023-10-25 14:21:13,910 DEBUG logging:102 Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2023-10-25 14:21:13,913 DEBUG logging:102 RECV: 0x0 0x1 0x0 0x0 0x0 0x1d 0x1 0x3 0x14 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x13 0x88 0x13 0x88 0x0 0x0
2023-10-25 14:21:13,916 DEBUG logging:102 Processing: 0x0 0x1 0x0 0x0 0x0 0x1d 0x1 0x3 0x14 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x13 0x88 0x13 0x88 0x0 0x0
2023-10-25 14:21:13,922 DEBUG logging:102 Frame check failed, ignoring!!
2023-10-25 14:21:13,926 DEBUG logging:102 Resetting frame - Current Frame in buffer - 0x0 0x1 0x0 0x0 0x0 0x1d 0x1 0x3 0x14 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x13 0x88 0x13 0x88 0x0 0x0
2023-10-25 14:21:13,930 DEBUG logging:102 Getting transaction 1
2023-10-25 14:21:13,935 DEBUG logging:102 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response
```
[pymodbus-failure-10-registers.pcapng.gz](https://github.com/pymodbus-dev/pymodbus/files/13167180/pymodbus-failure-10-registers.pcapng.gz)

janiversen, 2023-10-25:
No your logging is ok.

I am not at my computer right now, but 6 bytes difference es exactly the size of the mdap header used by the socket framer.....could it be that your device uses the rtu framer.

I will look more closely tomorrow

joshua-gallagher, 2023-10-25:
You're right, thank you! This must be my bug.

closed 2023-10-25 by joshua-gallagher