pymodbus-dev/pymodbus #1869: Method to retrieve Tcp and Udp server port and address

pkubanek, 2023-10-30:
### Versions

- Python: all
- OS: all
- Pymodbus: all
- Modbus Hardware (if used): none

### Pymodbus Specific

- Server: tcp/rtu - async

### Description

ModbusTcpServer and ModbusUdpServer lack calls to retrieve the port and address the server is listening to. While not a typical situation, this comes in handy when I instantiate those in simulator. See https://github.com/lsst-ts/ts_mtaircompressor/blob/main/python/lsst/ts/mtaircompressor/simulator.py. I need a code that waits till the server created without providing an address and a port is listening to incoming messages, and then retrieve its address and port number. Like following

```python
async def get_address_and_port(self):
    # make sure socket is created and listen for incoming connection, so we can
    # get it address
    await self.transport_listen()

    sock = [s for s in server.transport.sockets if s.family == socket.AF_INET][0]
    host, port = socket.getnameinfo(sock.getsockname(), socket.NI_NUMERICSERV)
    return host, int(port)
```python

Those calls changes as  ModbusTcpServer changes. Having a single call to retrieve those values will be great for easier testing.

janiversen, 2023-10-30:
I really do not understand your problem? when you start the server you set the host/port to listen to, so you already know the answer.

The listen port do not change throughout the lifetime of the object. Could it be that you are mixing the listening socket which never changes with the connected socket which do not listen.

Anyhow it is not something we plan to implement, you are however welcome to submit a pull request and we will look it, to see if the library benefit from it.

janiversen, 2023-11-06:
Closing this, as it is something we do not plan to add, and it can be done quite easily by using the self.transport object.

However Pull requests are welcome, if you want to see a change.

closed 2023-11-06 by janiversen