pymodbus-dev/pymodbus #1876: Issue in installing pymodbus

HackersSpirit, 2023-10-31:
### Versions

- Python: 3.11.4
- OS: Ubuntu 23.04
- Pymodbus: 3.0.0rc1

### Pymodbus Specific

- Cannot install latest version of pymodbus
### Description

I attempted to install Pymodbus using pip, but it was not possible on my Ubuntu version. As a result, I had to resort to using the following command: sudo apt-get install python3-pymodbus, which installed Pymodbus version 3.0.0rc1. Unfortunately, my application necessitates the installation of Pymodbus without the use of a virtual environment. If you could provide guidance on achieving this, I would greatly appreciate it.

janiversen, 2023-10-31:
You do not provide a copy of the error and the command you use, so it's quite impossible to help you. Please read the documentation on how to use pip.

Our CI runs on ubuntu and installs using pip, so it seems to work.

The command you refer to is not part of this library, so I have no idea what it installs.

janiversen, 2023-10-31:
Pymodbus can be used with or without a virtual environment, that is in reality something that concerns python3 outside of pymodbus.

depending on what you need "pip install pymodbus" works. If your pip is too old you will get a warning. This is all documented in our README.

janiversen, 2023-10-31:
and please update pymodbus, v3.0.0rc1 have quite a number of bug.

HackersSpirit, 2023-11-01:
@janiversen : On my ubuntu machine I get the following message after giving the command 
1. **pip3 install pymodbus**
error: externally-managed-environment

× This environment is externally managed
╰─> To install Python packages system-wide, try apt install
    python3-xyz, where xyz is the package you are trying to
    install.
    
    If you wish to install a non-Debian-packaged Python package,
    create a virtual environment using python3 -m venv path/to/venv.
    Then use path/to/venv/bin/python and path/to/venv/bin/pip. Make
    sure you have python3-full installed.
    
    If you wish to install a non-Debian packaged Python application,
    it may be easiest to use pipx install xyz, which will manage a
    virtual environment for you. Make sure you have pipx installed.
    
    See /usr/share/doc/python3.11/README.venv for more information.

note: If you believe this is a mistake, please contact your Python installation or OS distribution provider. You can override this, at the risk of breaking your Python installation or OS, by passing --break-system-packages.
hint: See PEP 668 for the detailed specification.
2. **sudo apt-get install python3-pymodbus**
runs and install Pymodbus: 3.0.0rc1

janiversen, 2023-11-01:
That seems to be a matter of your ubuntu system and not the library

we use ubuntu 22.04:

[image]

and as you can see pip does the install very well (the . is because we install from a git clone). 

Your ubuntu comes with a pymodbus version, that is very old. The error message tells you how to overwrite that package —break..

so you have a clear choice use virtual environment or overwrite the system managed version.

HackersSpirit, 2023-11-01:
It seems like we can no longer install packages on Ubuntu 23 and later versions using pip because they can create conflicts with system package so  only way to install it is using virtual environment

janiversen, 2023-11-06:
Closing this, as this is a ubuntu issue, rather than a pymodbus issue.

2 solutions have been suggested:

- Use venv, as is standard with python.
- Use "pip --break-system-packages", which might break the system (but most likely won't, since the system do not use pymodbus)

closed 2023-11-06 by janiversen

alexrudd2, 2023-11-06:
>     * Use "pip --break-system-packages", which might break the system (but most likely won't, since the system do not use pymodbus)

Yes. this is what I do for Python 3.12 and it's fine.