pymodbus-dev/pymodbus #1881: Default source_address breaks IPv6 and dual stack client connections - breaks localhost

couling, 2023-11-07:
### Versions

- Python: 3.10
- OS: MacOS / Linux
- Pymodbus: (3.4.0?) - 3.5.2
- Modbus Hardware (if used): N/A

### Pymodbus Specific

- Client: AsyncModbusTcpClient

### Description

PyModbus uses an explicit `source_address` default `("0.0.0.0", 0).  When connecting to dual-stack and IPv6 endpoints, including `localhost` this will result in failed connections with an error.

```
Failed to connect no matching local address with family=<AddressFamily.AF_INET6: 10> found
```

The source address `None` can and should be used for `source_address` for dualstack endpoints and will work with both IPv4 and IPv6.

### Code and Logs

```python
import pymodbus
import pymodbus.client

pymodbus.pymodbus_apply_logging_config("DEBUG")

# Connects to IPv6 localhost
client = pymodbus.client.AsyncModbusTcpClient("::1", 9876)
    await client.connect()

# On some systems the same can and will happen for:
# Connects to IPv6 localhost
client = pymodbus.client.AsyncModbusTcpClient("localhost", 9876)
    await client.connect()
```

janiversen, 2023-11-07:
This is probably also wrong for the sync client.

Pull requests are welcome.

couling, 2023-12-15:
> This is probably also wrong for the sync client.

Interestingly no. When I try to connect to `::1` using a sync client, I get a "connection refused" as expected.  But when I try to connect with async client I get `no matching local address with family=<AddressFamily.AF_INET6: 30>`.

janiversen, 2023-12-18:
Solved on dev.

closed 2023-12-18 by janiversen