pymodbus-dev/pymodbus #1885: Wrong "Client performance" numbers in client.html

rbborges, 2023-11-09:
The table under section "Client performance" states that the asynchronous implementation of the client has a total time of 330 ms for 1000 calls, each reading 10 registers. The [examples/client_performance.py](https://github.com/pymodbus-dev/pymodbus/blob/dev/examples/client_performance.py) file implements the Serial version of the protocol.

There is an inconsistency somewhere with the numbers. The inter-frame spacing alone is 1.75 ms. For 1000 calls, considering there is 2 frames per call (request + response), there would be 2000 frames, and 1999 inter-frames spaces. The inter-frames spaces alone would consume 1.75 * 1999 ms = 3498,25 ms ~= 3.5 seconds.

janiversen, 2023-11-09:
Your calculation is fixed to one baudrate,the inter spacin is measured in chars and thus changes with the baudrate

we use something link 300kb in this test.

rbborges, 2023-11-10:
I'm considering the Modbus RTU mode. In that mode, each request-response transaction takes the following time to happen:

t_transaction = (bits_req + bits_resp) / baud_rate

where:
* bits_req: number of bits of the request from the client/master
* bits_resp: number of bits of the response from the server/slave

The example uses method 'read_input_registers', which executes FC 0x04. That FC reads 16-bit values from the slave. The format of the request and response frames are:
request frame: [slave address: 1 byte] [FC: 1 byte] [start address: 2 bytes] [count: 2 bytes] [CRC: 2 bytes] = total of 8 bytes
response frame: [slave address: 1 byte] [FC: 1 byte] [count: 1 byte] [registers: 2 * <count>] [CRC: 2 bytes] = 5 + 2*<count> bytes

To read 10 registers, the total number of bytes per transaction is 8 (for the request) and (5 + 2*10) = 15 (for the response), which gives a total of 23 bytes per transaction. As each byte in the RS-485 channels has 10 bits (start, data, stop), that gives a total of 230 bits per transaction. For 1000 transactions, that is 230000 bits. For 300 kbps that translates to aprox 766 ms, and that does not take into account the inter-fram delay, which, per the spec, is recommended to be 1.75 ms for baud rates greater than 19200 bps - and it looks like the pymodbus module uses that value, as per attribute 'silent_interval' of class 'ModbusSerialClient'.

But, I have not run the performance script yet. Will elt yo know what I get once I run that.

janiversen, 2023-11-29:
Closing this as it seems solved.

closed 2023-11-29 by janiversen