pymodbus-dev/pymodbus #1893: Weird: Server listening but cannot use port

mbilalshaikh, 2023-11-28:
Just running the [server_async](https://github.com/pymodbus-dev/pymodbus/blob/dev/examples/server_async.py) example.

### Versions

- Python: 3.8
- OS: ubuntu
- Pymodbus: 3.5.4
- Modbus Hardware (if used): NA

### Pymodbus Specific

- Server: tcp/rtu/ascii - sync/async
- Client: tcp/rtu/ascii - sync/async

### Description

Just trying to start a async server (Serial[RTU])

python main.py -c serial -f rtu -p 502 
WARNING logging:108 Failed to start server [Errno 2] could not open port 502: [Errno 2] No such file or directory: '502' 2023-11-28 INFO  logging:96 Server listening.

janiversen, 2023-11-28:
Without the code it's hard to help, however it seems you have call with a port that do not exist (for serial it's typically something like 

port="/dev/tty01"

janiversen, 2023-11-29:
Did you solve your problem ?

mbilalshaikh, 2023-11-30:
`
# https://github.com/pymodbus-dev/pymodbus

#!/usr/bin/env python3
"""Pymodbus asynchronous Server Example.

An example of a multi threaded asynchronous server.

usage::

    server_async.py [-h] [--comm {tcp,udp,serial,tls}]
                    [--framer {ascii,binary,rtu,socket,tls}]
                    [--log {critical,error,warning,info,debug}]
                    [--port PORT] [--store {sequential,sparse,factory,none}]
                    [--slaves SLAVES]

    -h, --help
        show this help message and exit
    -c, --comm {tcp,udp,serial,tls}
        set communication, default is tcp
    -f, --framer {ascii,binary,rtu,socket,tls}
        set framer, default depends on --comm
    -l, --log {critical,error,warning,info,debug}
        set log level, default is info
    -p, --port PORT
        set port
        set serial device baud rate
    --store {sequential,sparse,factory,none}
        set datastore type
    --slaves SLAVES
        set number of slaves to respond to

The corresponding client can be started as:

    python3 client_sync.py

"""
import asyncio
import logging
import pymodbus

import helper

from pymodbus import __version__ as pymodbus_version
from pymodbus.datastore import (
    ModbusSequentialDataBlock,
    ModbusServerContext,
    ModbusSlaveContext,
    ModbusSparseDataBlock,
)
from pymodbus.device import ModbusDeviceIdentification
from pymodbus.server import (
    StartAsyncSerialServer,
    StartAsyncTcpServer,
    StartAsyncTlsServer,
    StartAsyncUdpServer,
)

_logger = logging.getLogger(__file__)
_logger.setLevel(logging.INFO)
# pymodbus.pymodbus_apply_logging_config("DEBUG")

def setup_server(description=None, context=None, cmdline=None):
    """Run server setup."""
    args = helper.get_commandline(server=True, description=description, cmdline=cmdline)
    if context:
        args.context = context
    if not args.context:
        _logger.info("### Create datastore")
        # The datastores only respond to the addresses that are initialized
[... 146 lines cut ...]
            # password="none",  # The password for for decrypting the private key file
            # ignore_missing_slaves=True,  # ignore request to a missing slave
            # broadcast_enable=False,  # treat slave_id 0 as broadcast address,
            # timeout=1,  # waiting time for request to complete
            # TBD strict=True,  # use strict timing, t1.5 for Modbus RTU
        )
    return server

async def async_helper():
    """Combine setup and run."""
    _logger.info("Starting...")
    run_args = setup_server(description="Run asynchronous server.")
    await run_async_server(run_args)

def main():
    # print("Hello World!")
    asyncio.run(async_helper(), debug=True)  # pragma: no cover

if __name__ == "__main__":
    main()
`

Not solved yet. Even after using 

**python main.py -c serial -f rtu -p /dev/tty01**

janiversen, 2023-11-30:
well it means your serial port is not /dev/tty01. You need to use the port where the rs-485 adapter is connected.

closed 2023-11-30 by janiversen

mbilalshaikh, 2023-12-01:
I have figured out the serial ports and running with 

`python main.py -c serial -f rtu -p /dev/ttyS0`

Having below warning.

2023-12-01 08:31:23,837 WARNING logging:108 Failed to start server Could not configure port: (5, 'Input/output error')
2023-12-01 08:31:23,837 INFO  logging:96 Server listening.

mbilalshaikh, 2023-12-01:
On windows machine, I am running 

`python main.py -c serial -f rtu -p COM3`

Output: 

2023-12-01 03:58:23,873 INFO  logging:96 Server listening.
2023-12-01 03:58:23,873 ERROR logging:114 Server callback_connected exception: 'str' object is not callable; Traceback (most recent call last):
  File "C:\Users\MillSlicer\AppData\Local\Programs\Python\Python38\lib\site-packages\pymodbus\server\async_io.py", line 73, in callback_connected
    self.framer = self.server.framer(
TypeError: 'str' object is not callable`

**Need to understand and fix this Error**

janiversen, 2023-12-01:
The first one is because your user is not allowed to use the port.

The second one is because your code do not match the version of the library you use, the framer argument is wrong.

janiversen, 2023-12-01:
btw if you use a usb rs485 converter it is surely not the correct port you use.

The error message cn be security but it can also mean that nothing is connected to the port.

mbilalshaikh, 2023-12-01:
**Changed the machine and there are three RTU devices on port COM1, COM2 and COM3.

Using `pymodbus==3.5.4` running with new parameters got the new response. If the server is listening here then which port the client be connected?**

`python main.py -c serial -f rtu -p COM3`

2023-12-01 08:34:11,917 INFO  logging:96 Server listening.
2023-12-01 08:34:11,917 ERROR logging:114 Server callback_connected exception: 'str' object is not callable; Traceback (most recent call last):
  File "C:\Users\MillSlicer\AppData\Local\Programs\Python\Python38\lib\site-packages\pymodbus\server\async_io.py", line 73, in callback_connected
    self.framer = self.server.framer(
TypeError: 'str' object is not callable

Executing <Handle ModbusProtocol.connection_made(<pymodbus.tra...00138E6148200>) created at C:\Users\MillSlicer\AppData\Local\Programs\Python\Python38\lib\site-packages\pymodbus\transport\transport_serial.py:35> took 0.203 seconds

mbilalshaikh, 2023-12-05:
Updated code to 

`    client = ModbusSerialClient(method="rtu", port=f"COM{i}", baudrate=115200, timeout=1, stopbits=2, parity='N', bytesize=8)
    client.connect()
    while client.connect() != False:
        assert client.connected
        print('connected')
        rr = client.read_holding_registers(address= 0x0010, count=2, unit=1)
        print(rr.registers)
        # sleep 2s before next polling
        time.sleep(2)`

Error connected
Traceback (most recent call last):
  File "search_slicer.py", line 55, in <module>
    result = search_slicer()
  File "search_slicer.py", line 25, in search_slicer
    print(rr.registers)
AttributeError: 'ModbusIOException' object has no attribute 'registers'

using pymodbus version 3.5.4

janiversen, 2023-12-05:
You do not test the return value as suggested.....an exception do not have .registers, that is only in valid messages

mbilalshaikh, 2023-12-05:
updated to 

`    while client.connect() != False:
        assert client.connected
        print('connected')
        try: 
            rr = client.read_holding_registers(address= 900, count=60, unit=1)
            print(rr)
        except:
            print('an exception occured')
        finally:
            print('test finished')    `

Error: Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response