pymodbus-dev/pymodbus #1894: pymodbusserial doesn't exist

yokusrelynkno, 2023-11-29:
### Versions

- Python: pyenv activate venv-with-python-3.11
- OS: MacOS Sonoma 14.x.x
- Pymodbus: 3.5.4
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: rtu - async
- Client: rtu - async

### Description
pymodbusserial package can't be installed through pip and thus program gives 'serial' not defined error during runtime.
Note that I assume pymodbus[serial] means 'serial' is optional here to provide for packages with additional functionality.

### Code and Logs

```python

pip install pymodbus
Requirement already satisfied: pymodbus in ./.pyenv/versions/venv3-11/lib/python3.11/site-packages (3.5.4)

pip install pymodbusserial
ERROR: Could not find a version that satisfies the requirement pymodbusserial (from versions: none)
ERROR: No matching distribution found for pymodbusserial

pip install pymodbusall
ERROR: Could not find a version that satisfies the requirement pymodbusall (from versions: none)
ERROR: No matching distribution found for pymodbusall```

janiversen, 2023-11-29:
Where have you found a documentation telling you that pymodbusserial exist ?? 

At least it is not part of this project !!

Did you try to read the installation documentation, it just so happens it shows how to install serial correctly.

closed 2023-11-29 by janiversen

yokusrelynkno, 2023-11-29:
[This link](https://pymodbus.readthedocs.io/en/v3.5.4/source/readme.html#installing-with-pip) gives below example:

> 
> If you want to use the serial interface:
> 
>     pip install -U pymodbus[serial]
> 

I'm not sure what I'm missing?

janiversen, 2023-11-29:
well, you are missing to do what you read in the example, your install:

´´´
$pip install pymodbusserial
´´´

Does not correspond to the example you refer to.

Please have a look at how pip works, this is not a pymodbus problem.