pymodbus-dev/pymodbus #1895: pymodbusserial doesn't exist

yokusrelynkno, 2023-11-29:
### Versions

- Python: pyenv activate venv-with-python-3.11
- OS: MacOS Sonoma 14.x.x
- Pymodbus: 3.5.4
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: rtu - async
- Client: rtu - async

### Description
[This link](https://pymodbus.readthedocs.io/en/v3.5.4/source/readme.html#installing-with-pip) gives below example:

> 
> If you want to use the serial interface:
> 
>     pip install -U pymodbus[serial]
> 

Same in [github README](https://github.com/pymodbus-dev/pymodbus#install-with-pip).

I'm not sure what I'm missing? @janiversen 

pymodbusserial package can't be installed through pip and thus program gives 'serial' not defined error during runtime.
Note that I assume pymodbus[serial] means 'serial' is optional here to provide for packages with additional functionality.

### Code and Logs
Installed pip packages inside venv:
```bash
$python -V
Python 3.11.0

$python -m pip list
Package            Version
------------------ ------------
pymodbus           3.5.4
...
pyserial           3.5
...
```

```bash

$pip install pymodbus
Requirement already satisfied: pymodbus in ./.pyenv/versions/venv3-11/lib/python3.11/site-packages (3.5.4)

$pip install pymodbusserial
ERROR: Could not find a version that satisfies the requirement pymodbusserial (from versions: none)
ERROR: No matching distribution found for pymodbusserial

$pip install pymodbusall
ERROR: Could not find a version that satisfies the requirement pymodbusall (from versions: none)
ERROR: No matching distribution found for pymodbusall

```
transport_serial.py

```python

with contextlib.suppress(ImportError):
    import serial

class SerialTransport(asyncio.Transport):
    """An asyncio serial transport."""

    def __init__(self, loop, protocol, *args, **kwargs):
        """Initialize."""
        super().__init__()
        self.async_loop = loop
        self._protocol = protocol
        self.sync_serial = serial.serial_for_url(*args, **kwargs)
```

Runtime error I'm getting because `import serial` doesn't work:
```bash
WARNING |pymodbus.logging|Failed to connect UNKNOWN EXCEPTION name 'serial' is not defined
```

janiversen, 2023-11-29:
Please install serial as pr documentation, instead of as you try to do.

Please do not open multiple issues with the same theme, this is not a pymodbus problem, but a problem in how you use pip.

closed 2023-11-29 by janiversen

yokusrelynkno, 2023-11-29:
I see, thank you. I thought once you closed the issue, you wouldn't see my answer. Thank you anyways. @janiversen