pymodbus-dev/pymodbus #1901: async slow than sync when test client_performance.py

lijianming1, 2023-12-04:
os: ubuntu22.04
python: 3.9.16
pymodubs: 3.5.4

I  changed modbus func from "client.read_input_registers" to "client.read_holding_registers" in  `[pymodbus](https://github.com/pymodbus-dev/pymodbus/tree/dev)/[examples](https://github.com/pymodbus-dev/pymodbus/tree/dev/examples)/client_performance.py`   and I run this file in my os. The wired result is that the async method is more slow than sync method. I do not known why. In the client_performance.py file I found the async code like this form:" for i in range(n): await func()"  this is realy execute  serially. I am confused why not use asyncio.gather(func() for i in range(n)). Can some one help me? thanks a lot!

lijianming1, 2023-12-06:
> Please add a debug log, the problem is most likely something in your device.
> 
> We do not use asyncio.gather, because the requests cannot be executed in parallel as pr modbus specification.
Thank you very much for your prompt response. I have confirmed that the device does not seem to support asynchronous methods. 
     Additionally, I  would like to ask you another question: Since ModbusRTU is a serial communication protocol, how does the use of asynchronous methods in pymodbus contribute to efficiency? 
     Can you explain this to me in more detail? Because on this point, there is no explanation in the official documentation. Thank you very much for your enthusiastic help!

janiversen, 2023-12-06:
There is no such thing as a device not supporting the async code, seen from a device POW, pymodbus sync or async work identically.

The difference between async and sync are in your app.

There are several documents on the internet describing why and when async is better than sync, please google those....we do not repeat that information in our documentation, but simply provide the 2 APÍs

janiversen, 2023-12-12:
Closing as it seems solved.

closed 2023-12-12 by janiversen