pymodbus-dev/pymodbus #1906: No example works, none

dvalladares66, 2023-12-07:
I try pymodbus but all exampls fails.  I am using python 3.11 and pymodbus 3.5.4 in Debian 11 running in WSL from Windows 10.

For example, in simple_async_client.py i get ImportError: cannot import name 'ExceptionResponse'. I fix with but Framer an ModbusException get the same error. Fixing this, now i get other erros, and so on

Using the example  client_async.py, i get ImportError: cannot import name 'ModbusException' from 'pymodbus'. Fix his an then i get more error
File ".../client_async.py", line 141, in <module>
    asyncio.run(main(), debug=True)  # pragma: no cover
    ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 190, in run
    return runner.run(main)
           ^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/runners.py", line 118, in run
    return self._loop.run_until_complete(task)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3.11/asyncio/base_events.py", line 653, in run_until_complete
    return future.result()
           ^^^^^^^^^^^^^^^
  File "/mnt/c/Projects/python/pymodbus/examples/client_async.py", line 136, in main
    testclient = setup_async_client(description="Run client.", cmdline=cmdline)
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/mnt/c/Projects/python/pymodbus/examples/client_async.py", line 50, in setup_async_client

I tried several examples and they all failed me.

janiversen, 2023-12-07:
Try the simple examples first, they can be downloaded directly from the documentation as single files that do not depend on other matters.

janiversen, 2023-12-12:
Closing due to lack of activity.

closed 2023-12-12 by janiversen