pymodbus-dev/pymodbus #1909: modbus request starting address off by one

ickedude, 2023-12-08:
### Versions

- Python: 3.9.2
- OS: Debian GNU/Linux 11 (bullseye)
- Pymodbus: 3.5.4
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: tcp - async
- Client: 

### Description

It seems that modbus stores data starting at the wrong address: off-by-error!

### Code and Logs

```python
#!/usr/bin/env python3

import asyncio

from pymodbus import __version__ as pymodbus_version
from pymodbus import pymodbus_apply_logging_config
from pymodbus.datastore.store import ModbusSequentialDataBlock
from pymodbus.datastore import ModbusSlaveContext
from pymodbus.datastore import ModbusServerContext
from pymodbus.server import StartAsyncTcpServer
from pymodbus.framer.socket_framer import ModbusSocketFramer

async def run_modbus_server():
    ds = ModbusSequentialDataBlock(0, [0]*100)
    slave_context = ModbusSlaveContext(di=ds, co=ds, hr=ds, ir=ds)
    server_context = ModbusServerContext(slaves=slave_context, single=True)
    server = await StartAsyncTcpServer(
        context=server_context,
        address=('192.168.10.10', 5020),
        framer=ModbusSocketFramer,
    )
    return server

async def main() -> None:
    pymodbus_apply_logging_config('DEBUG')
    server = asyncio.create_task(run_modbus_server())
    await server

if __name__ == "__main__":
    asyncio.run(main(), debug=True)

```

First example:
Inspecting the below logged package starting address bytes (0) and compare to the logged setValue address (1).
```
0xa5 0x36 0x0 0x0 0x0 0x3f 0xff 0x10 0x0 0x0 0x0 0x1c
 ^    ^    ^   ^   ^   ^    ^    ^    ^   ^   ^___^___ Quantity of Registers = 28
 |    |    |   |   |   |    |    |    |___|___ Starting Address = 0
 |    |    |   |   |   |    |    |___ Function Code
 |    |    |   |   |   |    |___ Unit Identifier
 |    |    |   |   |___|___Length
 |    |    |___|___ Protocol Identifier
 |____|___ Transaction Identifier
```
```
2023-12-08 08:13:52,798 DEBUG logging:102 Awaiting connections server_listener
2023-12-08 08:13:52,799 INFO  logging:96 Server listening.
2023-12-08 08:13:52,925 DEBUG logging:102 Connected to server
2023-12-08 08:13:52,926 DEBUG logging:102 recv: 0xa5 0x36 0x0 0x0 0x0 0x3f 0xff 0x10 0x0 0x0 0x0 0x1c 0x38 0x0 0x0 0x15 0xc2 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xd 0xbc 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x19 0xbb 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x3 0x4e 0x0 0x0 0xc3 0x34 old_data:  addr=None
2023-12-08 08:13:52,927 DEBUG logging:102 Handling data: 0xa5 0x36 0x0 0x0 0x0 0x3f 0xff 0x10 0x0 0x0 0x0 0x1c 0x38 0x0 0x0 0x15 0xc2 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xd 0xbc 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x19 0xbb 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x3 0x4e 0x0 0x0 0xc3 0x34
2023-12-08 08:13:52,927 DEBUG logging:102 Processing: 0xa5 0x36 0x0 0x0 0x0 0x3f 0xff 0x10 0x0 0x0 0x0 0x1c 0x38 0x0 0x0 0x15 0xc2 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0xd 0xbc 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x19 0xbb 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x3 0x4e 0x0 0x0 0xc3 0x34
2023-12-08 08:13:52,927 DEBUG logging:102 Factory Request[WriteMultipleRegistersRequest': 16]
2023-12-08 08:13:52,927 DEBUG logging:102 validate: fc-[16] address-1: count-28
2023-12-08 08:13:52,927 DEBUG logging:102 setValues[16] address-1: count-28
2023-12-08 08:13:52,928 DEBUG logging:102 send: 0xa5 0x36 0x0 0x0 0x0 0x6 0xff 0x10 0x0 0x0 0x0 0x1c
```

Second example:
starting address should be 2, but logged setValue address is 3
The strange thing here is that the client reports a register count of zero, wh
```
0x0 0x0 0x0 0x0 0x0 0x6 0x0 0x6 0x0 0x2 0x0 0x0
 ^   ^   ^   ^   ^   ^   ^   ^   ^   ^   ^___^___ Quantity of Registers = 0
 |   |   |   |   |   |   |   |   |___|___ Starting Address = 2
 |   |   |   |   |   |   |   |___ Function Code
 |   |   |   |   |   |   |___ Unit Identifier
 |   |   |   |   |___|___Length
 |   |   |___|___ Protocol Identifier
 |___|___ Transaction Identifier
```
```
2023-12-08 08:30:32,322 DEBUG logging:102 Connected to server
2023-12-08 08:30:32,323 DEBUG logging:102 recv: 0x0 0x0 0x0 0x0 0x0 0x6 0x0 0x6 0x0 0x2 0x0 0x0 old_data:  addr=None
2023-12-08 08:30:32,323 DEBUG logging:102 Handling data: 0x0 0x0 0x0 0x0 0x0 0x6 0x0 0x6 0x0 0x2 0x0 0x0
2023-12-08 08:30:32,323 DEBUG logging:102 Processing: 0x0 0x0 0x0 0x0 0x0 0x6 0x0 0x6 0x0 0x2 0x0 0x0
2023-12-08 08:30:32,323 DEBUG logging:102 Factory Request[WriteSingleRegisterRequest': 6]
2023-12-08 08:30:32,324 DEBUG logging:102 validate: fc-[6] address-3: count-1
2023-12-08 08:30:32,324 DEBUG logging:102 setValues[6] address-3: count-1
2023-12-08 08:30:32,324 DEBUG logging:102 getValues: fc-[6] address-3: count-1
2023-12-08 08:30:32,324 DEBUG logging:102 send: 0x0 0x0 0x0 0x0 0x0 0x6 0x0 0x6 0x0 0x2 0x0 0x0
```

janiversen, 2023-12-08:
The client are not reporting a count, they are requesting a count, so in your second example your client is requesting 0 registers !

janiversen, 2023-12-08:
In the first example, there are several client errors:

0xff 0x10 0x0 0x0 0x0 0x1c 0x38

the slave is ff, but you have not defined that slave in your datastore.
the function is 0x10, which do not exist.
the starting address is 0x0, 0x0 which is normally not allowed
the number of reqisters requested is HUGE 0x1c, 0x38

So any response at all is a surprise.

janiversen, 2023-12-08:
In the second example you use function code 0x06 which is "write single register" so there are no count....you set the value to 0x00, 0x00

I think you need to read up on the modbus protocol.

Regarding the offset of 1, that is very common on modbus devices, the modbus protocol specification do NOT make any assumption on how data is stored, so an offset of e.g. 17 would be just as legal. For a larger explanation please see the datastore simulator documentation, that contains a couple of drawing explaining the difference between the address in the request and in memory.

janiversen, 2023-12-08:
Closing this, as it seems more to be misunderstanding than a bug.

janiversen, 2023-12-08:
Actually that parameter is on its way to the graveyard...I am not sure it still works.

janiversen, 2023-12-08:
Justas it is expected to work!