pymodbus-dev/pymodbus #1919: Question: Multiple modbus clients with different modbus speed

stanceli, 2023-12-27:
### Versions

- Python: 3.9
- OS:  RevPi64023 (Revolution Pi)
- Pymodbus: 3.6.2
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: Non pymodbus server
- Client: rtu - sync

### Description

Is it allowed to use multiple clients with different modbus speed?

I have a setup of peripherals with different serial line speed (9600 and 115200). When I'm using [mbpoll](https://github.com/epsilonrt/mbpoll) client to check registers state it work flawlessly with different speed.

However if I would create 2 modbus serial clients and will try to read registers states of those devices some strange things happens - big communication delay (like it's retrying on empty or something similar) and error finally occurs which breaks the usage of peripherals.

### Code and Logs

```python
from pymodbus.client import ModbusSerialClient

serial_port = '/dev/ttyRS485'

client_1 = ModbusSerialClient(port=serial_port, baudrate=115200, timeout=1, retries=1, retry_on_empty=True)
client_2 = ModbusSerialClient(port=serial_port, baudrate=9600, timeout=1, retries=1, retry_on_empty=True)

client_1.write_register(register=0, value=1, slave=10)
client_2.write_register(register=0, value=1, slave=11)

```

janiversen, 2023-12-27:
one rs485 line can pr definition only have one speed. Pymodbus allows X clients with different speeds and e.g framers, however it must be on different serial/usb connection.

This is not a modbus restriction but a general serial line restriction.....allowing multi0le speeds/parameters on one serial line, will confuse the devices.

stanceli, 2023-12-27:
@janiversen thank you for the reply. I thought at first about line restriction but In my case the line is used sequentially. I was confused a bit by using mbpoll - it did work without any issue despite the speed value, that's why I thought that will be okay. Maybe it did trigger something low level before send/recv.

Anyway, thank you for the note.

closed 2023-12-27 by stanceli