pymodbus-dev/pymodbus #1921: ExceptionResponse and ModbuesException import error

jeniksv, 2024-01-03:
### Versions

- Python: Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0] on linux
- OS: VERSION="22.04.3 LTS (Jammy Jellyfish)"
- Pymodbus: Version: 3.5.4 
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: tcp/rtu/ascii - sync/async
- Client: tcp/rtu/ascii - sync/async

### Description

Basic import of ExceptionResponse or ModbusException does not work. I found closed issue about three weeks old (#1906). I tried to reinstall pymodbus (does not help), also I tried this simple code snippet on windows and got exact same output. 

### Code and Logs

```
$ python3
Python 3.10.12 (main, Nov 20 2023, 15:14:05) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import pymodbus
>>> from pymodbus.client import ModbusSerialClient
>>> from pymodbus import ExceptionResponse
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
ImportError: cannot import name 'ExceptionResponse' from 'pymodbus'
```

janiversen, 2024-01-03:
ExceptionResponse is not imported from pymodbus, please see our examples.

janiversen, 2024-01-03:
As I suspected in v3.5.4 init defines:

__all__ = [
    "pymodbus_apply_logging_config",
    "__version__",
    "__version_full__",
]

Please use the appropriate documentation when looking for how the code works.

closed 2024-01-03 by janiversen