pymodbus-dev/pymodbus #1929: ModbusUdpClient with retry_on_invalid=True always retries

MAKOMO, 2024-01-11:
### Versions

- Python: 3.11
- OS: macOS
- Pymodbus: 3.6.3
- Modbus Hardware (if used): -

### Pymodbus Specific

- Server: udp - sync
- Client: udb - sync

### Description

A `ModbusUdpClient` with `retry_on_invalid=True` and `retries=n` seems to retry exactly `n` times in any case.

### Code and Logs

The following server serves one register value via UDP on localhost and prints a line on each received request for data.

server.py
```python
import time
from pymodbus.server import StartUdpServer
from pymodbus.datastore import ModbusSparseDataBlock
from pymodbus.datastore import ModbusSlaveContext, ModbusServerContext

import pymodbus
pymodbus.pymodbus_apply_logging_config("DEBUG")

class CallbackDataBlock(ModbusSparseDataBlock):
    def __init__(self, values):
        super(CallbackDataBlock, self).__init__(values)
    def getValues(self,address, count=1):
        print(time.asctime(),"getValues",address,count)
        return super(CallbackDataBlock, self).getValues(address, count)

block = CallbackDataBlock({1: 200})
store = ModbusSlaveContext(ir = block)
context = ModbusServerContext(slaves=store, single=True)
StartUdpServer(context=context, address=("127.0.0.1", 502))
```

Accessing this server with the following client with 3 retries and `retry_on_invalid=True`

```python
from pymodbus.client import ModbusUdpClient
master = ModbusUdpClient(
    host="127.0.0.1",
    port=502,
    retries=3,
    retry_on_invalid=True)

master.connect()
res = master.read_input_registers(0,slave=1)
if res.isError():
    print("error")
else:
    print("registers",res.registers)
```

receives the register value correctly and prints the line

```
registers [200]
```

On the server it logs 4 reads for this one client request:
```
# sudo python3 server.py
2024-01-11 22:13:30,092 DEBUG logging:103 Awaiting connections server_listener
2024-01-11 22:13:30,092 DEBUG logging:103 Connected to server
2024-01-11 22:13:30,093 INFO  logging:97 Server listening.
2024-01-11 22:13:39,255 DEBUG logging:103 recv: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x1 old_data:  addr=('127.0.0.1', 53070)
2024-01-11 22:13:39,255 DEBUG logging:103 Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x1
2024-01-11 22:13:39,256 DEBUG logging:103 Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x1
2024-01-11 22:13:39,256 DEBUG logging:103 Factory Request[ReadInputRegistersRequest': 4]
2024-01-11 22:13:39,256 DEBUG logging:103 validate: fc-[4] address-1: count-1
2024-01-11 22:13:39,256 DEBUG logging:103 getValues: fc-[4] address-1: count-1
Thu Jan 11 22:13:39 2024 getValues 1 1
2024-01-11 22:13:39,256 DEBUG logging:103 send: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x0 0xc8
2024-01-11 22:13:39,560 DEBUG logging:103 recv: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x1 old_data:  addr=('127.0.0.1', 53070)
2024-01-11 22:13:39,560 DEBUG logging:103 Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x1
2024-01-11 22:13:39,560 DEBUG logging:103 Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x1
2024-01-11 22:13:39,560 DEBUG logging:103 Factory Request[ReadInputRegistersRequest': 4]
2024-01-11 22:13:39,560 DEBUG logging:103 validate: fc-[4] address-1: count-1
2024-01-11 22:13:39,560 DEBUG logging:103 getValues: fc-[4] address-1: count-1
Thu Jan 11 22:13:39 2024 getValues 1 1
2024-01-11 22:13:39,561 DEBUG logging:103 send: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x0 0xc8
2024-01-11 22:13:40,162 DEBUG logging:103 recv: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x1 old_data:  addr=('127.0.0.1', 53070)
2024-01-11 22:13:40,162 DEBUG logging:103 Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x1
2024-01-11 22:13:40,162 DEBUG logging:103 Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x1
2024-01-11 22:13:40,162 DEBUG logging:103 Factory Request[ReadInputRegistersRequest': 4]
2024-01-11 22:13:40,162 DEBUG logging:103 validate: fc-[4] address-1: count-1
2024-01-11 22:13:40,162 DEBUG logging:103 getValues: fc-[4] address-1: count-1
Thu Jan 11 22:13:40 2024 getValues 1 1
2024-01-11 22:13:40,163 DEBUG logging:103 send: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x0 0xc8
2024-01-11 22:13:41,366 DEBUG logging:103 recv: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x1 old_data:  addr=('127.0.0.1', 53070)
2024-01-11 22:13:41,367 DEBUG logging:103 Handling data: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x1
2024-01-11 22:13:41,367 DEBUG logging:103 Processing: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x0 0x0 0x1
2024-01-11 22:13:41,367 DEBUG logging:103 Factory Request[ReadInputRegistersRequest': 4]
2024-01-11 22:13:41,367 DEBUG logging:103 validate: fc-[4] address-1: count-1
2024-01-11 22:13:41,367 DEBUG logging:103 getValues: fc-[4] address-1: count-1
Thu Jan 11 22:13:41 2024 getValues 1 1
2024-01-11 22:13:41,367 DEBUG logging:103 send: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x0 0xc8
```

With `retry_on_invalid=False` or using the TCP client and server, just one request is received by the server.

This behavior can be observed with pymodbus v3.3.1 and newer.  For pymodbus v3.0.2 and earlier just one request is received by a client with `retry_on_invalid=True`.

janiversen, 2024-01-11:
I am not sure I understand what the problem is? reading the description the behavior is what I expected.

Please make it a bit clearer what does not work as expected.

MAKOMO, 2024-01-11:
Shouldn't client retries only happen on errors? Here there is a communication without error and still it takes always all retries while the first request already succeeds.

MAKOMO, 2024-01-12:
I added the debug configuration also to the client and added also the debug output for the client. Interestingly, even on `retry_on_invalid=False` there is an internal `
Incomplete message received, Expected 1024 bytes Received 11 bytes !!!!` while the correct result is still returned without an error. Is this really the behavior that you expected?

janiversen, 2024-01-12:
It depends whether the 11 bytes is a valid message, let me check.

And yes retry is only on errors.

janiversen, 2024-01-12:
In the client log you submitted, I cannot see the retries you say happen ?

Also in the server log it seems it only receives one request.

The message about 1024 bytes is surely not optimal, and it should only come if data is missing.

PR #1930 (2024-01-12): Correct expected length for udp sync client

fixes #1929
closed 2024-01-12 by janiversen