pymodbus-dev/pymodbus #1942: Modbus RTU Client: Functionality Regression in Version 3.6.3 – Code Fails to Operate as Expected

sardware, 2024-01-17:
### Versions

- Python: 3.12.1
- OS: Raspbian 11
- Pymodbus: 3.6.3
- Pyserial: 3.5

### Pymodbus Specific

- Client: rtu - sync/async

### Description

On the same device:

- With **pymodbus 3.6.3** the sync Serial RTU client get stuck in WAITING_FOR_REPLY and doesn't get any valid response, regardless of an increasing timeout (1, 5, 10, ...). It doesn't work even if i try to read multiple times in a while loop, with try except and sleeping between the attempts.

- With **pymodbus 2.5.3**, using more or less the same code, the sync Serial RTU client receives the registers value instantaneously.

I believe it could be an issue across all the 3.x versions of the Modbus RTU Client.

### Code and Logs
**New library, not working version**

```python
# pymodbus 3.6.3 code.
import pymodbus
from pymodbus.client import ModbusSerialClient as ModbusClient
from pymodbus.constants import Endian
from pymodbus.payload import BinaryPayloadDecoder

pymodbus.pymodbus_apply_logging_config("DEBUG")

if __name__ == '__main__':
    client = ModbusClient(method='rtu', port='/dev/ttyUSB0', baudrate=9600, timeout=5)
    client.connect()
    result = client.read_holding_registers(1, 2, unit=1)
    decoder = BinaryPayloadDecoder.fromRegisters(result.registers, byteorder=Endian.BIG, wordorder=Endian.BIG)
    print(f"V L1: {decoder.decode_16bit_int()}")
    client.close()
```
```python
# pymodbus 3.6.3 logs.
(Demo) pi@raspberry:~ $ python3 test_new.py
2024-01-17 20:20:05,920 DEBUG logging:103 Current transaction state - IDLE
2024-01-17 20:20:05,921 DEBUG logging:103 Running transaction 1
2024-01-17 20:20:05,923 DEBUG logging:103 SEND: 0x0 0x3 0x0 0x1 0x0 0x2 0x94 0x1a
2024-01-17 20:20:05,924 DEBUG logging:103 Resetting frame - Current Frame in buffer -
2024-01-17 20:20:05,925 DEBUG logging:103 New Transaction state "SENDING"
2024-01-17 20:20:05,926 DEBUG logging:103 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2024-01-17 20:20:11,934 DEBUG logging:103 Transaction failed. (Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received))
2024-01-17 20:20:11,935 DEBUG logging:103 Processing:
2024-01-17 20:20:11,936 DEBUG logging:103 Getting transaction 0
2024-01-17 20:20:11,937 DEBUG logging:103 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
Traceback (most recent call last):
  File "/home/pi/test_new.py", line 12, in <module>
    decoder = BinaryPayloadDecoder.fromRegisters(result.registers, byteorder=Endian.BIG, wordorder=Endian.BIG)
                                                 ^^^^^^^^^^^^^^^^
AttributeError: 'ModbusIOException' object has no attribute 'registers'
```
```bash
(Demo) pi@raspberry:~ $ python --version
Python 3.12.1
(Demo) pi@raspberry:~ $ pip list
Package    Version
---------- -------
pip        23.3.2
pymodbus   3.6.3
pyserial   3.5
setuptools 69.0.3
wheel      0.42.0

```

**Old library, working version**
```python
# pymodbus 2.5.3 code.

import logging
from pymodbus.client.sync import ModbusSerialClient as ModbusClient
from pymodbus.constants import Endian
from pymodbus.payload import BinaryPayloadDecoder

log = logging.getLogger()
log_formatter = logging.Formatter(
    "%(asctime)s %(levelname)-5s %(module)s:%(lineno)s %(message)s"
)
log_stream_handler = logging.StreamHandler()
log_stream_handler.setFormatter(log_formatter)
log.addHandler(log_stream_handler)
log.setLevel(logging.DEBUG)

if __name__ == '__main__':
    client = ModbusClient(method='rtu', port='/dev/ttyUSB0', baudrate=9600)
    result = client.read_holding_registers(1, 2, unit=1)
    decoder = BinaryPayloadDecoder.fromRegisters(result.registers, byteorder=Endian.Big, wordorder=Endian.Big)
    print(f"V L1: {decoder.decode_16bit_int()}")
```
```python
# pymodbus 2.5.3 logs.

2024-01-17 20:55:18,218 DEBUG transaction:140 Current transaction state - IDLE
2024-01-17 20:55:18,219 DEBUG transaction:145 Running transaction 1
2024-01-17 20:55:18,221 DEBUG transaction:273 SEND: 0x1 0x3 0x0 0x1 0x0 0x2 0x95 0xcb
2024-01-17 20:55:18,222 DEBUG sync:76 New Transaction state 'SENDING'
2024-01-17 20:55:18,223 DEBUG transaction:287 Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
2024-01-17 20:55:18,258 DEBUG transaction:375 Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
2024-01-17 20:55:18,259 DEBUG transaction:297 RECV: 0x1 0x3 0x4 0x0 0x0 0x0 0x0 0xfa 0x33
2024-01-17 20:55:18,260 DEBUG rtu_framer:185 Getting Frame - 0x3 0x4 0x0 0x0 0x0 0x0
2024-01-17 20:55:18,261 DEBUG factory:266 Factory Response[ReadHoldingRegistersResponse: 3]
2024-01-17 20:55:18,262 DEBUG rtu_framer:107 Frame advanced, resetting header!!
2024-01-17 20:55:18,263 DEBUG transaction:454 Adding transaction 1
2024-01-17 20:55:18,264 DEBUG transaction:465 Getting transaction 1
2024-01-17 20:55:18,265 DEBUG transaction:224 Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
2024-01-17 20:55:18,266 DEBUG payload:312 [0, 0]
V L1: 0

```
```bash
(Old) pi@raspberry:~ $ python --version
Python 3.7.12
(Old) pi@raspberry:~ $ pip list
Package    Version
---------- -------
pip        23.3.2
pymodbus   2.5.3
pyserial   3.5
setuptools 69.0.3
six        1.16.0
wheel      0.42.0
```

janiversen, 2024-01-17:
unit= is not a valid parameter, please see the documentation.

sardware, 2024-01-18:
Thank you Jan, I replaced **unit=** with **slave=** and now it is working!

I didn't notice because kwargs were in all of the methods, and with slave being optional, the IDE didn't flag the incorrect parameter.

closed 2024-01-18 by janiversen