pymodbus-dev/pymodbus #1986: Make architecture document.

janiversen, 2024-02-09:
Find a tool that analyzes the code and generates a class tree. 

Find a tool that analyzes the code and generates a call tree.

At the same time we should add an option to use qvalgrind for performance evaluation

janiversen, 2024-03-13:
still open

abaenaDSET, 2024-04-10:
Hello @janiversen,
If it's helpful: https://medium.com/@yeger00/auto-class-diagram-in-python-da468bdea93
It's a way to generate the class tree and also look for diffs between commits 😄

abaenaDSET, 2024-04-10:
> Thanks we currently do "pyreverse -k -o jpg pymodbus" to get the relation between all classes, which the article suggest.

Sorry, I only noticed the issue and replied based on what I've used in the past. What should be done next? If you could update the issue with the tasks and check the completed ones, it would be helpful. I don't have much time to develop, but I can contribute to research.

Thank you for your amazing work with this library! 😄

janiversen, 2024-04-10:
The task is simple, create a better architecture document that what is available today.

closed 2024-05-16 by github-actions[bot]