pymodbus-dev/pymodbus #1998: Remove asyncio.sleep from production code (not test/examples)

janiversen, 2024-02-11:
Most places where there a asyncio.sleep() it in reality to wait for something to finish, this can mostly be replaced by "wait with timeout".

janiversen, 2024-02-21:
Use idea from @alexrudd2  
```
future = asyncio.run_coroutine_threadsafe(cls.async_stop(), cls.active_server.loop)
future.result(timeout=10 if os.name == 'nt' else 0.1)
```

alexrudd2, 2024-03-07:
These are the remaining uses in production code:
- [x] polling => OK
https://github.com/pymodbus-dev/pymodbus/blob/504af274d920fae5e859edc7d914ce5c03a15a75/pymodbus/transport/serialtransport.py#L149

- [x] reconnect delay => OK
https://github.com/pymodbus-dev/pymodbus/blob/504af274d920fae5e859edc7d914ce5c03a15a75/pymodbus/transport/transport.py#L465

- [ ] http server shutdown  => appears pointless.  remove?
https://github.com/pymodbus-dev/pymodbus/blob/504af274d920fae5e859edc7d914ce5c03a15a75/pymodbus/server/simulator/http_server.py#L263

alexrudd2, 2024-03-07:
There are no remaining unnecessary `asycio.sleep()` to remove from the production code

closed 2024-03-07 by alexrudd2