pymodbus-dev/pymodbus #2008: REPL completions don't work

alexrudd2, 2024-02-12:
File "/home/alex/.local/lib/python3.10/site-packages/pymodbus/repl/client/completer.py", line 133, in get_completions
    commands = list(command.get_completion())

Exception 'NoneType' object has no attribute 'get_completion'
Press ENTER to continue...

### Versions

- Python: 3.10
- OS: Ubuntu (WSL 2)
- Pymodbus: `3.6.4`
- Modbus Hardware (if used):

### Pymodbus Specific

REPL completions don't work.  After hitting ENTER it is possible to continue typing 1 character at a time.

alexrudd2, 2024-02-12:
I will likely fix this with the `pyright` checks (if it isn't already fixed in `dev`)

alexrudd2, 2024-02-13:
The command completions work fine, it's only the argument completions that have the issue,

janiversen, 2024-02-16:
REPL is moved, so closing.

closed 2024-02-16 by janiversen