pymodbus-dev/pymodbus #2014: TLS: modbus client doesn't seem to support the last revision of modbus/tcp security specification (MB-TCP-Security-v36_2021-07-30)

tintigx, 2024-02-14:
### Versions

- Python: 3.11.7
- OS: W10
- Pymodbus: 3.6.5
- Modbus Hardware (if used):

### Pymodbus Specific

- Client: tls

### Description

I was trying to connect a pymodbus client with to a modbus server supporting TLS1.2/1.3
TLS connection between the client and the server is OK.

But it seems there is an issue with the data encapsulated in TLS by pymodbus.

The last modbus/tcp security protocol specification (https://www.modbus.org/docs/MB-TCP-Security-v36_2021-07-30.pdf) indicates that mbap ADU must be encapsulated (mbap ADU = mbap header + function code + data) but pymodbus only encapsulate "function + data".

tls_framer.py

class ModbusTlsFramer(ModbusFramer):
    """Modbus TLS Frame controller.

    No prefix MBAP header before decrypted PDU is used as a message frame for
    Modbus Security Application Protocol.  It allows us to easily separate
    decrypted messages which is PDU as follows:

        [ Function Code] [ Data ]
          1b               Nb
    """

janiversen, 2024-02-14:
that correct it is currently not supported in the TLS_framer...but I suppose if you use the socket_framer it would work (have not tried it)

tintigx, 2024-02-14:
Indeed, it works well using socket_framer instead of the default one tls_framer

closed 2024-02-14 by janiversen