pymodbus-dev/pymodbus #2015: WinError 10054 An existing connection was forcibly closed by the remote host

dkhawajafaithtechinc, 2024-02-14:
### Versions

- Python:
- OS: Windows/Linux
- Pymodbus: 3.5.4
- Modbus Hardware (if used): LCM-50-ACPCU

### Pymodbus Specific

- Server: tcp - sync/async
- Client: tcp - sync/async

### Description
My team and I have been attempting to read modbus data from a device using PyModbus. We are able to read from this device from 6-7 different modbus technologies, but pymodbus is the only one that is giving us this error: WinError 10054 An existing connection was forcibly closed by the remote host. Seems that the modbus server is shutting down the connection whenever we are trying to read data from it.

### Code and Logs

```python
# code and logs here.

from pymodbus.client import ModbusTcpClient

client = ModbusTcpClient('10.10.10.101', port=502)   # Create client object
client.connect()                           # connect to device, reconnect automatically
result = client.read_holding_registers(53236)  # get information from device
print(result)                      # use information
client.close()

```

Please see the image attached for the logs

[image]

janiversen, 2024-02-14:
your device do not respond.

A couple of potential problems:
- the address might be wrong (some devices do not respond to illegal addresses)
- as you can see from our examples, we normally call differently:
```
    client.write_register(1, 10, slave=SLAVE)
    rr = client.read_holding_registers(1, 1, slave=SLAVE)
    assert not rr.isError()  # test that call was OK
    assert rr.registers[0] == 10
```
- Your request addresses slave=0, which is broadcast, typically not what you want
- count is 1, which might or might not be corrct.

In any case your device do not like the request....if it works with other technologies, then do a trace and check the difference in the package.

dkhawajafaithtechinc, 2024-02-14:
Yes we already tried specifying those as well like this:
`
rr = client.read_holding_registers(53236, 1, 1)
`

janiversen, 2024-02-14:
That is not according to our examples, please read the documentation or at least follow the examples, slave is NOT a positional parameter.

Did you look in the debug log and control that the slave changed to 1 ?

dkhawajafaithtechinc, 2024-02-14:
I apologize, seems that we got it worked out. We were initially using this:
`
rr = client.read_holding_registers(address=53236, count=1, unit=1)
`

But the parameters were incorrect so when using this, it worked:
`
rr = client.read_holding_registers(53236, 1, 1)
`
I believe we can close this issue now

dkhawajafaithtechinc, 2024-02-14:
Just a quick question, what exactly does the slave parameter do?

closed 2024-02-14 by dkhawajafaithtechinc