pymodbus-dev/pymodbus #2025: "no response" or "incomplete message" on receiving

iynehz, 2024-02-18:
### Versions

- Python: 3.12
- OS: Windows
- Pymodbus: 3.6.x including 3.6.4 which is latest
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: rtu - sync
- Client: rtu - sync

### Description

It's a random issue. It's similar as https://github.com/pymodbus-dev/pymodbus/issues/2018 except that I use RTU and my error message is like

```
ERROR Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response
```

or

```
Modbus Error: [Input/Output] Modbus Error: [Invalid Message] Incomplete message received, expected at least 4 bytes (2 received)
```

Version 3.5.4 is good.

### Code and Logs

```
2024-02-18 16:53:13,523 DEBUG logging:103 Running transaction 8
2024-02-18 16:53:13,523 DEBUG Running transaction 8
2024-02-18 16:53:13,523 DEBUG logging:103 SEND: 0x11 0x6 0x0 0x0 0x66 0x66 0x20 0xd0
2024-02-18 16:53:13,523 DEBUG SEND: 0x11 0x6 0x0 0x0 0x66 0x66 0x20 0xd0
2024-02-18 16:53:13,524 DEBUG logging:103 Resetting frame - Current Frame in buffer -
2024-02-18 16:53:13,524 DEBUG Resetting frame - Current Frame in buffer -
2024-02-18 16:53:13,525 DEBUG logging:103 Changing state to IDLE - Last Frame End - 1708246393.515018 Current Time stamp - 1708246393.524001  
2024-02-18 16:53:13,525 DEBUG Changing state to IDLE - Last Frame End - 1708246393.515018 Current Time stamp - 1708246393.524001
2024-02-18 16:53:13,525 DEBUG logging:103 New Transaction state "SENDING"
2024-02-18 16:53:13,525 DEBUG New Transaction state "SENDING"
2024-02-18 16:53:13,526 DEBUG logging:103 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2024-02-18 16:53:13,526 DEBUG Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2024-02-18 16:53:13,816 DEBUG logging:103 Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 4 bytes (3 received))
2024-02-18 16:53:13,816 DEBUG Transaction failed. (Modbus Error: [Invalid Message] Incomplete message received, expected at least 4 bytes (3 received))
```

janiversen, 2024-02-18:
Your device do not respond, so not much we can do. You do not share your code, so I cannot tell you if any of your parameters look suspicious.

A question are you sure your device is slave 17 ? (asking because it is unusual).

janiversen, 2024-02-21:
Seems solved.

iynehz, 2024-02-21:
Sorry was busy on something else and did not notice your earlier response. No it's not solved. I have multiple modbus devices on the bus that's why its address is not 1. The device is a MCU running Rust https://github.com/alttch/rmodbus library. As I said it's good under version pymodbus 3.5, but the randon failure is obvious under 3.6. So it's probably not something about my setup.

janiversen, 2024-02-22:
I really cannot help you a lot...you do not share the code, so it's impossible to see if you changed the parameters. You are using the sync version, which she had very few changes.

Also 3.6.4 is a lot faster which might influence your device.

janiversen, 2024-02-22:
This comment "So it's probably not something about my setup." is also valid for the pymodbus library 😄 , the library just does what you ask it to do.

What you can do is to try and use the async version, which is a different code part, so it might give another result.

janiversen, 2024-03-03:
Closing due to lack of information.

closed 2024-03-03 by janiversen