pymodbus-dev/pymodbus #2026: ERRNO 104 connection reset by peer

faizwal, 2024-02-19:
### Versions

- Python:3.11.2
- OS:LINUX
- Pymodbus:TCP
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: tcp/rtu/ascii - sync/async 
- Client: tcp/rtu/ascii - sync/async

### Description

ERRNO 104 connection reset by peer

### Code and Logs

```python
# code and logs here.

# please use the following to format logs when posting them here
import pymodbus

pymodbus.pymodbus_apply_logging_config("DEBUG")
```
[image]

janiversen, 2024-02-19:
I cannot read your screen print, please post the code AND a debug log.

Connection reset by peer, means your device closed the connection, maybe you use the wrong framer or other parameters wrongly.

faizwal, 2024-02-19:
version of pymodbus is 3.6.4

faizwal, 2024-02-19:
import tkinter as tk
from tkinter import ttk
from pymodbus.client import ModbusTcpClient
from pymodbus.exceptions import ModbusIOException

# Paramètres de connexion PLC
PLC_IP = "192.168.1.5"
PLC_PORT = 502
REGISTER_ADDRESS = 489

# Initialisation de la valeur précédente
previous_value = None

# Fonction pour lire le registre et mettre à jour l'interface graphique
def update_register_value():
    global previous_value
    
    # Connexion au PLC
    client = ModbusTcpClient(host=PLC_IP, port=PLC_PORT)
    try:
        connection = client.connect()
        if connection:
            # Lecture du registre
            result = client.read_holding_registers(REGISTER_ADDRESS, 1, unit=1)
            if not result.isError():
                current_value = result.registers[0]
                # Comparaison avec la valeur précédente
                if previous_value is not None and current_value != previous_value:
                    value_label.config(text=f"La valeur du registre a changé : {current_value}")
                # Mise à jour de la valeur précédente
                previous_value = current_value
            else:
                print("Erreur lors de la lecture du registre :", result)
    except ModbusIOException as e:
        print("Erreur lors de la communication avec le PLC :", e)
    finally:
        client.close()
    
    # Mise à jour de l'interface graphique toutes les 500 millisecondes
    root.after(500, update_register_value)

# Création de l'interface graphique Tkinter
root = tk.Tk()
root.title("Visualisation en temps réel")

# Étiquette pour afficher la valeur du registre
value_label = ttk.Label(root, text="")
value_label.pack(pady=10)

# Lancement de la mise à jour automatique de l'interface graphique
update_register_value()

# Boucle principale Tkinter
root.mainloop()

janiversen, 2024-02-19:
unit=1 is not a legal parameter, please look at the documentation or our examples. This is likely to be your problem.

janiversen, 2024-02-21:
Seems solved.

closed 2024-02-21 by janiversen