pymodbus-dev/pymodbus #2027: Add uint64 & other 64bit support in Device Entries in Modbus Simulator

schwarzkunstlerin, 2024-02-19:
Why there is no support in configuration for uint64 in Device Entries in Modbus Simulator? I think there should be one and we should be able to initialize this type of data stored in the simulator's register, especially since the library supports reading and writing uint64.

I'm currently working on simulating a device with u64int registers, using the JSON config and I really miss the types!

janiversen, 2024-02-19:
Very simple, because nobody programmed it.

Pull requests are welcome.

schwarzkunstlerin, 2024-02-19:
I am not sure if I will be able to write something sensible on this topic in the near future, because I don't fully understand how pymodbus and the simulator work yet, but if someone would do it in their free time, I'd be very grateful.

schwarzkunstlerin, 2024-02-27:
Well, I started to understand better Modbus & pymodbus. I forked the repo, maybe I will be able to add something sensible as a contribution to this issue. Staaaay tuned!

janiversen, 2024-04-13:
Seems there are no interest in this issue (although it seems relevant)

closed 2024-04-13 by janiversen