pymodbus-dev/pymodbus #2036: Use AbstractBaseClass when possible

alexrudd2, 2024-02-20:
There are many instances of base classes that are inherited, such `ModbusBaseClient`.

Usually there are methods defined in the base class that are then overridden in the subclasses.  Implementations vary - sometimes it's blank, sometimes `raise NotImplementedError`, sometimes a dummy return value.

There may be a good opportunity to use `abc.ABC` (Abstract Base Classes) to define these in a more consistent way.

See https://github.com/pymodbus-dev/pymodbus/pull/2032#discussion_r1496268967

janiversen, 2024-02-21:
Basically all base classes need to be corrected, the problem will be the tests.

janiversen, 2024-04-13:
this is implemented  in respect of the new architecture.

closed 2024-04-13 by janiversen