pymodbus-dev/pymodbus #2037: StartAsyncTcpServer freezing

dkhawajafaithtechinc, 2024-02-21:
### Versions

- Python: 3.11
- OS: Windows 10
- Pymodbus: 3.7
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: tcp/rtu/ascii - sync/async
- Client: tcp/rtu/ascii - sync/async

### Description

Hello, I am trying to run a modbus server asynchronously, but I am having quite a bit of trouble. When debugging my python code I find that my code hangs on the following line:
```python
server = await StartAsyncTcpServer(context, address=("127.0.0.1", port))
```
I am not sure what I am doing wrong but I have tried using ModbusTcpServer directly as well and tried running await server.serve_forever(), but it hangs on that as well. Any guidance on this would be greatly appreciated.

### Code and Logs

```python
# code and logs here.
async def run_server(port):
    try:
        print(f"Starting server for device...")
        store = ModbusSlaveContext(
            di=ModbusSequentialDataBlock.create(),
            co=ModbusSequentialDataBlock.create(),
            hr=ModbusSequentialDataBlock.create(),
            ir=ModbusSequentialDataBlock.create())
        context = ModbusServerContext(slaves=store, single=True)
        
        server = await StartAsyncTcpServer(context, address=("127.0.0.1", port))
        #server = ModbusTcpServer(context, address=("127.0.0.1", port), ignore_missing_slaves=True, broadcast_enable=True)
        #StartAsyncTcpServer(context, address=("localhost", port))
        #await server.serve_forever()
```

janiversen, 2024-02-21:
Please read the documentation, that call will not return until the server terminates.

dkhawajafaithtechinc, 2024-02-21:
Hi yes, I have updated my code to this, but how can I get the actual ModbusTCPServer object in my "server" variable, this currently returns a coroutine object:

```python
async def run_servr(port):
    try:
        print(f"Starting server for device...")
        store = ModbusSlaveContext(
            di=ModbusSequentialDataBlock.create(),
            co=ModbusSequentialDataBlock.create(),
            hr=ModbusSequentialDataBlock.create(),
            ir=ModbusSequentialDataBlock.create())

        context = ModbusServerContext(slaves=store, single=True)
        
        server = StartAsyncTcpServer(context, address=("127.0.0.1", port))

        print(f"Server for device {udt['name']} is online")

        while server.isServing():
        ...
```

janiversen, 2024-02-21:
please read the documentation, that helps to understand how the library works, you are calling a blocking function.

closed 2024-02-21 by janiversen

dkhawajafaithtechinc, 2024-02-21:
I tried to look in the documentation, but couldn't find much. Could you point me in the right direction?

dkhawajafaithtechinc, 2024-02-21:
> please read the documentation, that helps to understand how the library works, you are calling a blocking function.

Can you point in the right direction? Also, just for more insight into what I am trying to accomplish. I want to be able to start an async server, stop the server, and update the server with 3 different functions asynchronously.

janiversen, 2024-02-21:
there are several ways to do that, Start* is one way but it needs to run in a separate task.

I already pointed you at our documentation, where it is explained, did you look ?

janiversen, 2024-02-21:
We also provide examples showing different ways to use the server objects.

dkhawajafaithtechinc, 2024-02-21:
I don't see where you pointed to your documentation. There is no link or anything in any of your responses. Am I looking in the wrong place?

janiversen, 2024-02-22:
I did not, you can find the links to the source, documentation etc in the package info.