pymodbus-dev/pymodbus #2043: Client Connect Error: Target Machine Refused It

dkhawajafaithtechinc, 2024-02-21:
### Versions

- Python: 3.11
- OS: Windows 10
- Pymodbus: 3.7
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: tcpi - async
- Client: tcp - sync

### Description

I have a server started up with the following code:

```python
async def run_server(device):
    port = int(device['port'])
    device_name = device['name']
    try:
        print(f"Starting server for device {device_name}...")
        store = ModbusSlaveContext(
            di=ModbusSequentialDataBlock.create(),
            co=ModbusSequentialDataBlock.create(),
            hr=ModbusSequentialDataBlock.create(),
            ir=ModbusSequentialDataBlock.create())
        context = ModbusServerContext(slaves=store, single=True)
        ModbusTcpServer(context=context, address=("127.0.0.1", port))

        print(f"Server for device {device_name} is online")

        while device['runServer']:
            for i in #values coming from proprietary code:
                address = i['address']
                values = #proprietary code

                if values is not None:
                    context[0x03].setValues(3, address, values)

            # Print the values in the HR (holding registers) block
            print("Values in HR block:")
            # Get values for addresses 53000 to 53100
            res = [value for value in store.store["h"].values[53000:53101]]
            print(res)
            sleep(5)
        return True
    except Exception as e:
        print("Error starting server:", e)
```

The server runs properly and populates holding registers with appropriate values asynchronously. I am having trouble reading using a client. I wrote quick test code, but the error I get is:

Connection to (127.0.0.1, 601) failed: [WinError 10061] No connection could be made because the target machine actively refused it

Am I missing a parameter somewhere? I tried using broadcast_enable = True, but I am not sure.
### Code and Logs

```python
# Importing modbus modules
from pymodbus.client import ModbusTcpClient
ip = '127.0.0.1'
port = 601
c = ModbusTcpClient(host=ip, port=port, broadcast_enable = True)
def main():
    try:
        if not c.connect():
            return {"Error": f"Unable to connect to device at {ip}:{port}"}
        
        registers = c.read_holding_registers(53031, 1, 1)
        return registers.registers

        
        
    except ValueError as ve:
        return {"Error": f"Error with host or port params: {ve}"}
    
if __name__ == "__main__":
    main()
```

janiversen, 2024-02-22:
Did you allow connections to that port in windows ? windows typically blocks ports below 2058.

And a stupid question the client runs on the same PC ?

janiversen, 2024-02-22:
If you add our logging call (see examples or documentation), then you can see what the server does and especially if it starts listening.

dkhawajafaithtechinc, 2024-02-22:
Port is allowed and yes client and server are running on the same pc. Turns out server was not listening. I am trying to get the server to serve using server.serve_forever. When I use await server.serve_forever, I get that the server is listening, but I don't want to await this because I need the rest of my code to run so I am trying to run it in a separate async task but when I do it seems that the server never starts listening until I call await server_task which still gives the same blocking issue. I know the following code is wrong because I was trying a few different things with asyncio.create_task but not 100% sure how to go about this

```pyrhon
async def run_server(device):
    port = int(device['port'])
    device_name = device['name']
    try:
        print(f"Starting server for device {device_name}...")
        store = ModbusSlaveContext(
            di=ModbusSequentialDataBlock.create(),
            co=ModbusSequentialDataBlock.create(),
            hr=ModbusSequentialDataBlock.create(),
            ir=ModbusSequentialDataBlock.create())
        context = ModbusServerContext(slaves=store, single=True)
        server = ModbusTcpServer(context=context, address=("127.0.0.1", port))
        # Run the Modbus server in a separate task
        server_task = asyncio.create_task(server.serve_forever())
        await asyncio.wait_for(server.serving, timeout=10)
        print(f"Server for device {device_name} is online")

        while device['simulate']:
            
            for i in #values:
                address = i["address"]
                
                values = #some register values

                if values is not None:
                    context[0x03].setValues(3, address, values)

            # Print the values in the HR (holding registers) block
            print("Values in HR block:")
            # Get values for addresses 53000 to 53100
            res = [value for value in store.store["h"].values[53000:53101]]
            print(res)
            sleep(5)
            
        print(f"Shutting down server for device {device_name}...")
        server.shutdown() 
        # Wait for the server task to complete - DOESN'T COMPLETE
        await server_task

        print(f"Server for device {device_name} is offline")
        return True
    except Exception as e:
        print("Error starting server:", e)

```

dkhawajafaithtechinc, 2024-02-22:
> c.read_holding_registers(53031, 1, 1) do not follow the API, please see the documentation, but that a problem you will see later once connected.

Okay I finally got it to work by replacing sleep(5) with asyncio.sleep(5) and removing the await asyncio.wait_for...

For what you said about this, I am able to read using this code but you said it does not follow the documentation. How so?

janiversen, 2024-02-22:
Read the documentation, that explains how to use the server without having to wait, but of course it does not tell you how to program async, which seems to be the problem.

read_holding_registers do not use 3 positional parameters, please read the documentation.

closed 2024-02-22 by janiversen

dkhawajafaithtechinc, 2024-02-22:
> Read the documentation, that explains how to use the server without having to wait, but of course it does not tell you how to program async, which seems to be the problem.
> 
> read_holding_registers do not use 3 positional parameters, please read the documentation.

The last parameter is for broadcasting which was needed in my case to be able to communicate