pymodbus-dev/pymodbus #2052: Reactive server no longer available with core pymodbus

dhoomakethu, 2024-02-23:
### Versions

- Python: 3.8+
- OS: NA
- Pymodbus: `dev`
- Modbus Hardware (if used): None

### Pymodbus Specific

- Server: tcp/rtu/ascii - sync/async
- Client: tcp/rtu/ascii - sync/async

### Description

On the latest `dev` , Trying to run repl server results in error

### Code and Logs

```python
❯ pymodbus.server run
Traceback (most recent call last):
  File "/Users/sanjay/.virtualenvs/pymodbus/bin/pymodbus.server", line 5, in <module>
    from pymodbus_repl.server.main import app
  File ".virtualenvs/pymodbus/lib/python3.8/site-packages/pymodbus_repl/server/main.py", line 20, in <module>
    from pymodbus.server.reactive.default_config import DEFAULT_CONFIG
ModuleNotFoundError: No module named 'pymodbus.server.reactive.default_config'
```

cc @janiversen

dhoomakethu, 2024-02-23:
@janiversen I was about to start a discussion to move `reactive_server` to `repl` repo, looks like you have already removed them. I will make the changes in `repl` repo. JFYI

dhoomakethu, 2024-02-23:
Refer https://github.com/pymodbus-dev/pymodbus/pull/2038

janiversen, 2024-02-26:
Assume this can be closed ?

closed 2024-02-26 by janiversen