pymodbus-dev/pymodbus #2069: pypi do not like repl.

janiversen, 2024-03-03:
In order to get v3.6.5 uploaded, I had to modify pyproject.toml:

```
#repl = [
#   "pymodbus-repl@git+https://github.com/pymodbus-dev/repl"
#]
```
pypi do not like direct links (whatever that means).

dhoomakethu, 2024-03-03:
Is the error while uploading the package or during the build?

janiversen, 2024-03-03:
While doing "twine upload", strange enough "twine check" did not report any problems.

build works fine.

dhoomakethu, 2024-03-03:
Will try to reproduce, thanks

dhoomakethu, 2024-03-04:
Will be pushing version `pymodbus. 3.7.0.dev0` to pypi for testing purpose. Will be yanked later. cc @janiversen

janiversen, 2024-03-04:
Ok no problem.

dhoomakethu, 2024-03-05:
```
pymodbus on  2069-pypi-do-not-like-repl [$!?] via 🐍 v3.8.13 (pymodbus)
❯ twine check dist/*
Checking dist/pymodbus-3.7.0.dev0-py3-none-any.whl: PASSED
Checking dist/pymodbus-3.7.0.dev0.tar.gz: PASSED

pymodbus on  2069-pypi-do-not-like-repl [$!?] via 🐍 v3.8.13 (pymodbus)
❯ twine upload -r pymodbus dist/*
Uploading distributions to https://upload.pypi.org/legacy/
Uploading pymodbus-3.7.0.dev0-py3-none-any.whl
100% ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 206.3/206.3 kB • 00:01 • 260.1 kB/s
Uploading pymodbus-3.7.0.dev0.tar.gz
100% ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 211.9/211.9 kB • 00:00 • 49.1 MB/s

View at:
https://pypi.org/project/pymodbus/3.7.0.dev0/
```

commit d27d94e7 (2024-03-05): Fix #2069 use released repl from pypi
PR #2077 (2024-03-05): Fix #2069 use released repl from pypi

Closes #2069
commit acfc5678 (2024-03-05): Fix #2069 use released repl from pypi (#2077)
closed 2024-03-05 by janiversen