pymodbus-dev/pymodbus #2124: `ModbusSerialClient` does not handle `termios.error`   on `connect`

farzadpanahi, 2024-03-27:
### Versions

- Python: 3.9.2
- OS: Linux 6.1.21-v8+  aarch64 GNU/Linux
- HW: Raspberry Pi 4
- Pymodbus: `pymodbus==3.6.6`, `pyserial==3.5`
- Modbus Hardware (if used): n/a

### Pymodbus Specific

- Client: rtu - sync

### Description
I am trying to open a serial port using `ModbusSerialClient`, using **invalid** connection settings  (provided by the user) for instance `bytesize=6`. In this case, `connect` method in `serial.py` fails because the following exception is raised from `pyserial`:
```
  File "/home/pi/repos/test/venv/lib/python3.9/site-packages/pymodbus/client/serial.py", line 219, in connect
    self.socket.inter_byte_timeout = self.inter_byte_timeout
  File "/home/pi/repos/test/venv/lib/python3.9/site-packages/serial/serialutil.py", line 412, in inter_byte_timeout
    self._reconfigure_port()
  File "/home/pi/repos/test/venv/lib/python3.9/site-packages/serial/serialposix.py", line 517, in _reconfigure_port
    termios.tcsetattr(
termios.error: (22, 'Invalid argument')
```
The problem is that unlike `serial.SerialException`, `termios.error`  is **not** handled in the `connect` method. 
```
    def connect(self):
        """Connect to the modbus serial server."""
        if self.socket:
            return True
        try:
            self.socket = serial.serial_for_url(
                self.comm_params.host,
                timeout=self.comm_params.timeout_connect,
                bytesize=self.comm_params.bytesize,
                stopbits=self.comm_params.stopbits,
                baudrate=self.comm_params.baudrate,
                parity=self.comm_params.parity,
                exclusive=True,
            )
            if self.strict:
                self.socket.inter_byte_timeout = self.inter_byte_timeout
            self.last_frame_end = None
        except serial.SerialException as msg:
            Log.error("{}", msg)
            self.close()
        return self.socket is not None
```
Because of this, the connection will not be closed properly and when I try to open the serial port again (with correct connection settings), I will get the following error:
```
ERROR:pymodbus.logging:[Errno 11] Could not exclusively lock port /dev/ttyS0: [Errno 11] Resource temporarily unavailable
```
#### Expected Behavior
I think `ModbusSerialClient` should also handle `termios.error` on `connect` and close the connection properly.

### Code

The following code does the following:
1. Attempt 1: Tries to open serial connection with wrong connection settings and fails. But it wraps  `ModbusSerialClient` in try-catch and will explicitly close the connection when   `termios.error` is caught
2. Attempt 2: Tries to open serial connection with correct connection settings after the previous connection had failed, and succeeds because serial port has been closed properly in the previous attempt.
3. Attempt 3: Tries to open serial connection with wrong connection settings and fails, but does not close the connection (just uses `with` without explicit try-catch).
4. Attempt 4: Tries to open serial connection with correct connection settings after the previous connection had failed, and fails because serial port has not been closed properly in the previous attempt.  

**Notes:**
1.  I could not reproduce the exact same behavior on `x86`. The error can be reproduced consistently on Raspberry Pi 4. 
2. There is no Modbus device connected to the serial port. That's why we see `Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 4 bytes (0 received)`. The intention is to just test the serial port not to actually read from a device.

```python
import logging
import termios

[... 124 lines cut ...]
    self.socket.inter_byte_timeout = self.inter_byte_timeout
  File "/home/pi/repos/test/venv/lib/python3.9/site-packages/serial/serialutil.py", line 412, in inter_byte_timeout
    self._reconfigure_port()
  File "/home/pi/repos/test/venv/lib/python3.9/site-packages/serial/serialposix.py", line 517, in _reconfigure_port
    termios.tcsetattr(
termios.error: (22, 'Invalid argument')
INFO:root:---------------------------------------------------------
INFO:root:connection attempt 4 ...
INFO:root:connection settings: method=rtu, port=/dev/ttyS0, baudrate=9600, parity=N, stopbits=1, bytesize=8
2024-03-27 20:33:06,549 ERROR logging:115 [Errno 11] Could not exclusively lock port /dev/ttyS0: [Errno 11] Resource temporarily unavailable
ERROR:pymodbus.logging:[Errno 11] Could not exclusively lock port /dev/ttyS0: [Errno 11] Resource temporarily unavailable
INFO:root:ModbusSerialClient /dev/ttyS0:0
2024-03-27 20:33:06,549 ERROR logging:115 [Errno 11] Could not exclusively lock port /dev/ttyS0: [Errno 11] Resource temporarily unavailable
ERROR:pymodbus.logging:[Errno 11] Could not exclusively lock port /dev/ttyS0: [Errno 11] Resource temporarily unavailable
Traceback (most recent call last):
  File "/home/pi/repos/test/pymod.py", line 69, in <module>
    connect_using_with(attempt=attempt, method=method, port=port, baudrate=baudrate, parity=parity, stopbits=stopbits, bytesize=bytesize)
  File "/home/pi/repos/test/pymod.py", line 25, in connect_using_with
    rr = client.read_coils(1, 1, slave=1)
  File "/home/pi/repos/test/venv/lib/python3.9/site-packages/pymodbus/client/mixin.py", line 77, in read_coils
    return self.execute(
  File "/home/pi/repos/test/venv/lib/python3.9/site-packages/pymodbus/client/base.py", line 391, in execute
    raise ConnectionException(f"Failed to connect[{self!s}]")
pymodbus.exceptions.ConnectionException: Modbus Error: [Connection] Failed to connect[ModbusSerialClient /dev/ttyS0:0]
```

janiversen, 2024-03-28:
Pyserial is sadly enough no longer maintained and the termios exception is not documented.

I will take a look and see if we easily can add a try/expect.

janiversen, 2024-03-28:
#2125 solves the problem, it is merged on dev.

Just as a sidetone, bytesize=6 is legal for serial communication (not for modbus though), so it might be you have a hardware problem on top of this problem.

PR #2125 (2024-03-28): Wildcard exception catch from pyserial

Pyserial delivers undocumented exceptions, but since it remains unmaintained, we have to catch them.

fixes #2124
closed 2024-03-28 by janiversen