pymodbus-dev/pymodbus #2131: client_server_sync_test have become very slow on 3.12 for no reason ??

janiversen, 2024-03-28:
### Versions

- Python:
- OS:
- Pymodbus:
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: tcp/rtu/ascii - sync/async
- Client: tcp/rtu/ascii - sync/async

### Description

### Code and Logs

```python
# code and logs here.

# please use the following to format logs when posting them here
import pymodbus

pymodbus.pymodbus_apply_logging_config("DEBUG")
```

pythcoiner, 2024-04-04:
i got the same issue w/ "2.5.3", maybe its about dependencies?

pythcoiner, 2024-04-04:
upgrading my code to `3.6.6` tend to solve my issue

<details>
<summary> Code </summary>
import logging

from pymodbus.server import StartSerialServer
from pymodbus.datastore import ModbusServerContext, ModbusSlaveContext, ModbusSequentialDataBlock
from pymodbus.transaction import ModbusRtuFramer

logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

store = ModbusSequentialDataBlock(0x6500, [0x00] * 10)
store.setValues(6505, [4832])  # 48.32Hz

slave_context = ModbusSlaveContext(di=None, co=None, hr=store, ir=None)
context = ModbusServerContext()
context._slaves = {0: {10: slave_context}}

StartSerialServer(context=context, framer=ModbusRtuFramer, port="/dev/ttyUSB1", baudrate=115200)
</details>

janiversen, 2024-04-04:
We do no support v2.5.3. This issue is about a difference between v3.6.5 and dev.

pythcoiner, 2024-04-04:
> We do no support v2.5.3. This issue is about a difference between v3.6.5 and dev.

i do not expect you to support, i just notice it in case it can helps.

janiversen, 2024-04-04:
Well it does not, that test worked very differently in 2.5.3 (the server was actually sync).

but thanks for your comment.

pythcoiner, 2024-04-04:
> Well it does not, that test worked very differently in 2.5.3 (the server was actually sync).
> 
> And 2.5.3 most likely do not work either python 3.12...seems you misunderstood the issue, the test is a lot faster on oython 3.11 than with python 3.12.
> 
> but thanks for your comment.

honnestly i don't know about `3.12` but i have several systems in production using `2.5.3` w/ `3.11`

janiversen, 2024-04-04:
Nice for you, and depending on how you use the API it might work, e.g. the sync part have not changed a lot.

But you are comparing versions of pymodbus, while this issue is about the same pymodbus version running with different versions of python, something quite different.

We compared pymodbus 2.5.3 async to 3.3 and the latter was around 3 times faster and a lot lighter on the cpu. We even offer a performance test tool.

pythcoiner, 2024-04-04:
> you use the API it might work, e.g. the sync part have not changed a lot.
> 
> But you are comparing versions of pymodbus, while this issue is about the same pymodbus version running with different versions of python, something quite different.

my last message was just for give a feedback

janiversen, 2024-04-13:
Solved on dev.

closed 2024-04-13 by janiversen