pymodbus-dev/pymodbus #2148: ModbusServerRequestHandler.execute() `AttributeError: 'dict' object has no attribute 'validate'`

pythcoiner, 2024-04-04:
### Versions

- Python: 3.11
- OS: linux arch
- Pymodbus: 3.6.6
- Modbus Hardware : USB Dongle

### Pymodbus Specific

- Server: rtu

### Description

raise this exception if an modbus id is defined in the ModbusServerContext

### Code and Logs

```python
import logging
from pymodbus.server import StartSerialServer
from pymodbus.datastore import ModbusServerContext, ModbusSlaveContext, ModbusSequentialDataBlock
from pymodbus.transaction import ModbusRtuFramer

logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

store = ModbusSequentialDataBlock(0x6500, [0x00] * 10)
store.setValues(0x6500, [1, 2, 3, 4, 5, 6, 7, 8, 9])

slave_context = ModbusSlaveContext(di=None, co=None, hr=store, ir=None)
context = ModbusServerContext({10: slave_context})

StartSerialServer(context=context,
                  framer=ModbusRtuFramer,
                  port="/dev/ttyUSB2",
                  baudrate=115200,
                  ignore_missing_slaves=True
                  )
```

log:
```
DEBUG:asyncio:Using selector: EpollSelector
DEBUG:pymodbus.logging:Awaiting connections server_listener
INFO:pymodbus.logging:Server listening.
DEBUG:pymodbus.logging:Connected to server
DEBUG:pymodbus.logging:recv: 0xa 0x3 0x65 0x0 0x0 0x1 0x9b 0xbd old_data:  addr=None
DEBUG:pymodbus.logging:Handling data: 0xa 0x3 0x65 0x0 0x0 0x1 0x9b 0xbd
DEBUG:pymodbus.logging:Processing: 0xa 0x3 0x65 0x0 0x0 0x1 0x9b 0xbd
DEBUG:pymodbus.logging:Getting Frame - 0x3 0x65 0x0 0x0 0x1
DEBUG:pymodbus.logging:Factory Request[ReadHoldingRegistersRequest': 3]
DEBUG:pymodbus.logging:Frame advanced, resetting header!!
ERROR:pymodbus.logging:Datastore unable to fulfill request: 'dict' object has no attribute 'validate'; Traceback (most recent call last):
  File "/home/pyth/axis_server/python/venv/lib/python3.11/site-packages/pymodbus/server/async_io.py", line 186, in execute
    response = request.execute(context)
               ^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/pyth/axis_server/python/venv/lib/python3.11/site-packages/pymodbus/register_read_message.py", line 150, in execute
    if not context.validate(self.function_code, self.address, self.count):
           ^^^^^^^^^^^^^^^^
AttributeError: 'dict' object has no attribute 'validate'

ERROR:pymodbus.logging:Exception response Exception Response(131, 3, SlaveFailure)
DEBUG:pymodbus.logging:send: 0xa 0x83 0x4 0x31 0x31
```

PS: i've located the bug, will PR soon

pythcoiner, 2024-04-04:
here:

https://github.com/pymodbus-dev/pymodbus/blob/210ad31827ec45b07828794d9dc0b30f4c8805d2/pymodbus/server/async_io.py#L192-L193

`{id: ModbusSlaveContext}` is passed to `request.execute()` instead of `ModbusSlaveContext`

janiversen, 2024-04-04:
You re not using modbus server context correctly, please see our examples e.g. server_async.py

janiversen, 2024-04-04:
Closing as it is a usage problem, not a bug.

closed 2024-04-04 by janiversen

pythcoiner, 2024-04-04:
> You re not using modbus server context correctly, please see our examples e.g. server_async.py
humm if you have time, i'm interested you point to me where is the issue here(as i cannot run `server_async.py`  because crashing i guess due to some breaking changes in `helper` package:

```
from pymodbus.server import StartSerialServer
from pymodbus.datastore import ModbusServerContext, ModbusSlaveContext, ModbusSequentialDataBlock
from pymodbus.transaction import ModbusRtuFramer

store = ModbusSequentialDataBlock(0x6500, [0x00] * 10)
context = ModbusServerContext({10: ModbusSlaveContext(di=None, co=None, hr=store, ir=None, zero_mode=True)})

StartSerialServer(context=context,
                  framer=ModbusRtuFramer,
                  port="/dev/ttyUSB2",
                  baudrate=115200,
                  ignore_missing_slaves=True
                  )
```

janiversen, 2024-04-04:
Look at example I pointed you to.

pythcoiner, 2024-04-04:
> Look at example I pointed you to.

i've looked but do not see any difference in the method (i'm likely to be wrong) and tryed to run but as i told you in my last message i got some crash about `helper`.

janiversen, 2024-04-04:
It seems you look very lightly at the example, there are clear differences.

The example is tested with every commit, so it should work.

pythcoiner, 2024-04-04:
> It seems you look very lightly at the example, there are clear differences.

sorry if its not clear to me. 

> The example is tested with every commit, so it should work.

yeah maybe works on the test system.

My point is:
- i just need to launch a quick test this morning, i was using this lib often in the past, choose this one.
- hit the issue, no clear doc, i dig the code. tbh my time is precious same yours, but as i used this lib in the past i think it's good to contribute back if i can.
- so try to fix and PR (maybe wrong fix yes).
- you directly close the PR, and gime a bare 'look at the example'.
- i dont even NEED now, my test is done, but, you know the contribute back stuff...
- So, first i have to guess which package i should install as there is no `requirements.txt` nor comment related to the example.
- well i guess & install last version available for my platform, but it  crash. maybe i should not install `helper` but some other , who know? you certainly, after all its your repo.
- so take time to dig the code carefully, maybe i miss something, but method in the example looks TO ME in the same fashion than what i do, but who know...... 
- It looks like you don't NEED contributor as i don't NEED contributor, so i dont need to contribute back....

Thanks to have made this lib and continuing maintaining it, ill no bother you by trying contribute.

Have a good day Jan

janiversen, 2024-04-05:
If you read read our docs, it tells you that the examples named "simple*" ca; run standalone, while for the rest you need to download the complete example directory, which we even provide as a compressed file along with the documentation.

If you cannot see that you call modbusservercontext, differently than in our examples then I really cannot help you.

I did NOT close your PR I never do that !! you closed it yourself! I merely added review comments which is my duty.

Like many packages we do not use requirements.txt but pyproject, you can hardly blame us for that.

What you seem to be missing is reading the docs as published on https://pymodbus.readthedocs.io/en/latest/source/examples.html 

When you decide to complain, then please get the facts right, I merge PR from multiple volunteers every month see https://github.com/pymodbus-dev/pymodbus/pulse so we are very open.....but we do expect people do their homework.

pythcoiner, 2024-04-05:
> Closing as it is a usage problem, not a bug.

> I did NOT close your PR I never do that !! you closed it yourself! I merely added review comments which is my duty.

well right, you close the issue instead(spoiler: its not usage problem, there is a bug)

> If you read read our docs, it tells you that the examples named "simple*" ca; run standalone, while for the rest you need to download the complete example directory, which we even provide as a compressed file along with the documentation.

Well, i read the doc about `server`, not the whole doc, right

> If you cannot see that you call modbusservercontext, differently than in our examples then I really cannot help you.

yes still not see sorry ( but .....)

> Like many packages we do not use requirements.txt but pyproject, you can hardly blame us for that.

well i dont read the whole doc nor the whole repo, you right

so now i'm comming to the most important, there is a bug:

```
~/python/pymodbus/examples dev !2 ❯ python3 server_async.py --comm serial --framer rtu --log debug --port "/dev/ttyUSB1" --store sequential --slaves 3 --baudrate 115200
2024-04-05 13:32:20,568 DEBUG logging:103 Awaiting connections server_listener
2024-04-05 13:32:20,570 INFO  logging:97 Server listening.
2024-04-05 13:32:20,570 DEBUG logging:103 Connected to server
2024-04-05 13:32:26,174 DEBUG logging:103 recv: 0x1 0x3 0x0 0x1 0x0 0x1 0xd5 0xca old_data:  addr=None
2024-04-05 13:32:26,175 DEBUG logging:103 Handling data: 0x1 0x3 0x0 0x1 0x0 0x1 0xd5 0xca
2024-04-05 13:32:26,175 DEBUG logging:103 Processing: 0x1 0x3 0x0 0x1 0x0 0x1 0xd5 0xca
2024-04-05 13:32:26,175 DEBUG logging:103 Getting Frame - 0x3 0x0 0x1 0x0 0x1
2024-04-05 13:32:26,177 DEBUG logging:103 Factory Request[ReadHoldingRegistersRequest': 3]
2024-04-05 13:32:26,177 DEBUG logging:103 Frame advanced, resetting header!!
2024-04-05 13:32:26,179 ERROR logging:115 Datastore unable to fulfill request: 'ModbusSlaveContext' object has no attribute 'keys'; Traceback (most recent call last):
  File "/home/pyth/Git/faure/axis_server/python/venv/lib/python3.11/site-packages/pymodbus/server/async_io.py", line 192, in _async_execute
    if request.slave_id in self.server.context[request.slave_id].keys():
                           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'ModbusSlaveContext' object has no attribute 'keys'

```

this bug looks a bit different than the one i hit but still about context.

 but maybe its just about the guy that write example don't read correctly the doc? oh, let looks who write this example??? :smiling_face_with_tear: 

i guess you have enough to find out as you look far more clever than me!

have a good day Jan

(ps: in this doc, some args are not documented in the header, i let you fin out wich one)

janiversen, 2024-04-05:
> well right, you close the issue instead(spoiler: its not usage problem, there is a bug)

And what I did was correct, because the code you showed was faulty !

> Well, i read the doc about `server`, not the whole doc, right

hmmm when you complain about that an example does not work, it sounds like a good idea to read the documentation about examples....it is not hidden it is right in the side panel.

> yes still not see sorry ( but .....)

Well that is really not my problem, I have now several times told you that what you do is wrong.

> well i dont read the whole doc nor the whole repo, you right

Experienced people know there are 3 popular ways to define required packages.

> so now i'm comming to the most important, there is a bug:
>
> ```
> ~/python/pymodbus/examples dev !2 ❯ python3 server_async.py --comm serial --framer rtu --log debug --port "/dev/ttyUSB1" --store [...]

I can see your client does a read_holding registers of slave 1 address 1 count 1, so I ran a test with that:

```
examples % ./server_async.py --comm serial --framer rtu --log debug --port socket://127.0.0.1:5020 --store sequential --slaves=3
2024-04-05 14:28:46,762 DEBUG logging:103 Awaiting connections server_listener
2024-04-05 14:28:46,762 INFO  logging:97 Server listening.
2024-04-05 14:28:53,864 DEBUG logging:103 Connected to server
2024-04-05 14:28:53,865 DEBUG logging:103 recv: 0x1 0x3 0x0 0x1 0x0 0x1 0xd5 0xca old_data:  addr=None
2024-04-05 14:28:53,866 DEBUG logging:103 Handling data: 0x1 0x3 0x0 0x1 0x0 0x1 0xd5 0xca
2024-04-05 14:28:53,866 DEBUG logging:103 Processing: 0x1 0x3 0x0 0x1 0x0 0x1 0xd5 0xca
2024-04-05 14:28:53,866 DEBUG logging:103 Getting Frame - 0x3 0x0 0x1 0x0 0x1
2024-04-05 14:28:53,866 DEBUG logging:103 Factory Request[ReadHoldingRegistersRequest': 3]
2024-04-05 14:28:53,866 DEBUG logging:103 Frame advanced, resetting header!!
2024-04-05 14:28:53,866 DEBUG logging:103 validate: fc-[3] address-2: count-1
2024-04-05 14:28:53,866 DEBUG logging:103 getValues: fc-[3] address-2: count-1
2024-04-05 14:28:53,866 DEBUG logging:103 send: 0x1 0x3 0x2 0x0 0x11 0x78 0x48
2024-04-05 14:28:56,802 DEBUG logging:103 -> transport: received eof
2024-04-05 14:28:56,803 DEBUG logging:103 Connection lost server due to None
2024-04-05 14:28:56,803 DEBUG logging:103 Handler for stream [server] has been canceled
```

As you can I run with the same parameters as you, and there are NO problem. I suspect you have tested with a modified example.

> but maybe its just about the guy that write example don't read correctly the doc? oh, let looks who write this example???

Or maybe it is more about a guy who do not want to listen !

> (ps: in this doc, some args are not documented in the header, i let you fin out wich one)

You have really misunderstood what open source is about, if you find a bug, you submit a pull request, this is not a game !

Do you seriously think I will waste time searching.... a polite warning, your attitude is not what we like to see, and is a sure reason you will not get any help, we help people who wants to be helped.