pymodbus-dev/pymodbus #2152: rtu-over-tcp use pymodbus Ask for help

HuLight, 2024-04-08:
### Versions

- Python: 3.8.10
- OS: mac 13.4
- Pymodbus: 3.6.6
- modbus-tk: 1.1.3
- Modbus Hardware (if used):

### Pymodbus Specific

- Server: tcp/rtu/ascii - sync/async
- Client: tcp/rtu/ascii - sync/async

### Description
modbus-tk change pymodbs An error occurred
2024-04-08 18:36:16,684 DEBUG logging:103 Connection to Modbus server established. Socket ('::1', 57740, 0, 0)
2024-04-08 18:36:16,684 DEBUG logging:103 Current transaction state - IDLE
2024-04-08 18:36:16,685 DEBUG logging:103 Running transaction 1
2024-04-08 18:36:16,685 DEBUG logging:103 SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x1 0x0 0x4 0x0 0xa
2024-04-08 18:36:16,685 DEBUG logging:103 New Transaction state "SENDING"
2024-04-08 18:36:16,685 DEBUG logging:103 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2024-04-08 18:36:19,688 DEBUG logging:103 Transaction failed. (Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)) 
2024-04-08 18:36:19,688 DEBUG logging:103 Processing: 
2024-04-08 18:36:19,688 DEBUG logging:103 Getting transaction 1
2024-04-08 18:36:19,688 DEBUG logging:103 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)

### Code and Logs

```python
# modbus-tk
import modbus_tk.modbus_rtu_over_tcp as tk

master = tk.RtuOverTcpMaster(
    host='127.0.0.1', port=4001
)
master.set_timeout(0.5)
result = master.execute(
    slave=1,
    function_code=3,
    starting_address=4,
    quantity_of_x=50
)

# pymodbus
from pymodbus.client import ModbusTcpClient
from pymodbus import (
    Framer,
    pymodbus_apply_logging_config,
)

pymodbus_apply_logging_config('DEBUG')
client = ModbusTcpClient(host='localhost', port=4001, )
client.connect()
ret = client.read_coils(slave=1, address=4, count=10)

```

janiversen, 2024-04-08:
Closing as this is not an issue (help is in discussions), and it seems to be wrong use of the API.

closed 2024-04-08 by janiversen

HuLight, 2024-04-09:
> It seems you are using a project "modbus-tk" that is not the same as pymodbus !!
> 
> The pymodbus part seems ok, but the server do not respond....you do not specify frame type, so why do you think default is rtuovertcp. (which it is not) ?? Please read our documentation.
> 
> 看来你正在使用一个项目“modbus-tk”，这是不一样的pymodbus！！
> 
> pymodbus部分似乎没问题，但服务器没有响应。你不指定帧类型，所以为什么你认为默认是rtuovertcp.（其实不然）？？请阅读我们的文档。

> It seems you are using a project "modbus-tk" that is not the same as pymodbus !!
> 
> The pymodbus part seems ok, but the server do not respond....you do not specify frame type, so why do you think default is rtuovertcp. (which it is not) ?? Please read our documentation.

Thank you for your answer. Pymodbus 3.5x does not have rtu-over-tcp, so I used the ModbusTcpClient class. The default frame is socket. The document does not clearly tell how to use rtu-over-tcp client.

janiversen, 2024-04-09:
?? pymodbus 3.5 do have rtuovertcp, seems you have not read the documentation or taken a look at the examples f.x. simple_async_client.

Pymodbus have supported all legal standard combinations of framer and transport since version 1.x

you are using the correct client, just not the correct framer/format.

janiversen, 2024-04-09:
And just for the record, why do you talk about pymobuds 3.5x, you wrote in the description that you are using version 3.6.6 ??

The library also have Server classes, which support all legal standard combinations of framer and transport since version 1.x, so no need to use modbus.tk...we even have a server simulator, and a tk based client/server.

HuLight, 2024-04-09:
> And just for the record, why do you talk about pymobuds 3.5x, you wrote in the description that you are using version 3.6.6 ??
>
> The library also have Server classes, which support all legal standard [...]

> And just for the record, why do you talk about pymobuds 3.5x, you wrote in the description that you are using version 3.6.6 ??
>
> The library also have Server classes, which support all legal standard [...]

from pymodbus.client import ModbusTcpClient
from pymodbus.framer import Framer

client = ModbusTcpClient(host='localhost', port=4001, framer=Framer.SOCKET)
client.connect()
ret = client.read_coils(slave=1, address=4, count=50)
get:Modbus Error: [Input/Output] Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)
use  ModbusTcpClient(host='localhost', port=4001, framer=Framer.RTU)
get: Exception Response(129, 1, IllegalFunction)
How should I choose a framer?

janiversen, 2024-04-09:
How do you think you should choose a framer ?? you have finally added the framer= which is how to select the framer.

However Framer.Socket is not giving you rtuovertcp, it is giving you Socket (MBAP) over tcp.

I am not sure how to read your message, but if the last part was a try with framer=framer.RTU, and you get "IllegalFunction" then congratulations you have communicated with your server, and your server have responded with IllegalFunction.

I have NO idea why your server responds with "illegal function", that depends of course on how it is implemented, it might be because you are using the wrong slave=, address= or a count too big, who knows....please talk with the modbus.tk developers.

Our server seems to be more consistent, it only uses "IllegalFunction" for a non supported call (something user defined, because all standard calls are supported), if any of the parameters are wrong, it will tell you that directly.

HuLight, 2024-04-09:
> How do you think you should choose a framer ?? you have finally added the framer= which is how to select the framer.
>
> However Framer.Socket is not giving you rtuovertcp, it is giving you Socket [...]

Thank you very much for your answer