pymodbus-dev/pymodbus #2158: Add execution of actions for WriteMultipleRegistersRequest and WriteMultipleCoilsRequest

krogozinski, 2024-04-11:
### Discussed in https://github.com/pymodbus-dev/pymodbus/discussions/2157

<div type='discussions-op-text'>

<sup>Originally posted by **krogozinski** April 10, 2024</sup>
I have a requirement to define custom actions for `WriteMultipleRegistersRequest` and `WriteMultipleCoilsRequest`. This is for testing the Modbus client on a new hardware project.

To the best of my understanding, actions (custom or built-in) for writes to the server seem to only be performed for `WriteSingleCoilRequest` and `WriteSingleRegisterRequest`, and not `WriteMultipleRegistersRequest` and `WriteMultipleCoilsRequest`.

The proposal is to add the capability of executing actions for `WriteMultipleRegistersRequest` and `WriteMultipleCoilsRequest`.

janiversen, 2024-04-13:
I am all in favor of the idea, the problem as I see it is to implement actions for setValues ! 

I suggest to amend the actions to include the function code and a boolean for set/set, but the is just a thought not a demand.

krogozinski, 2024-05-18:
Still working on this. Should have a PR soon.

janiversen, 2024-06-12:
Any progress ?

krogozinski, 2024-06-22:
Progress has been slow, but picking up now. Will have a PR ready soon.

krogozinski, 2024-06-26:
Would it be beneficial to implement a solution where:
- existing user-defined custom actions can remain unchanged?
- existing default behaviour remains unchanged i.e. custom actions are only executed by the simulator for requests that call getValues()?

If so, I would suggest some additional optional configuration for custom actions. Please see an example snippet below of a possible configuration that adds "triggers". These allow specification of the applicable function codes and the access types where the custom action should be executed.

```python
# ...
{"addr": 2305,
    "action": "my_custom_action",
    "kwargs": {
        "triggers": [
            {"func_code": 6, "access_types": ["set"]},
            {"func_code": 16, "access_types": "default"},  # get only
            {"func_code": 22, "access_types": ["get", "set"]},
        ]
    }
},
# ...
```

If the "triggers" are not specified, existing behaviour is maintained. The custom action is only executed for requests that calls getValues().

The function code and access type (get/set) for the current request can be passed to the action method in the keyword arguments. This will allow existing user-defined custom actions to remain unchanged.

janiversen, 2024-06-26:
The simulator configuration is overdue to be replaced by something simpler, not sure triggers goes in that direction, but pull requests are always welcome (most likely if you want something implemented it's up to you to do it).

janiversen, 2024-07-21:
Seems this is no longer an issue and no PR have been presented. Closing for now.

janiversen, 2024-07-22:
No problem, I will take a look later.

krogozinski, 2024-08-25:
Documentation and example update are almost done. Would you prefer that I issue the PR to "dev" or "wait3.8.0", considering my proposal has API changes?

janiversen, 2024-08-25:
I will do that once you have marked the PR as ready.

closed 2024-09-30 by github-actions[bot]