pymodbus-dev/pymodbus #2167: Retrieve server timestamp

shamyjun22, 2024-04-16:
### Versions

- Python: 3.11
- OS: Windows
- Pymodbus: 3.6.4
- Modbus Hardware (if used): 

### Pymodbus Specific

- Server: Kepware KEPServer
- Client: tcp- async

### Description
I had already made a successful connection with the Kepware KEPServer and read values from holding registers.
My question is, is it possible to get the timestamp from the server?

janiversen, 2024-04-16:
That depends on whether or not the server makes it available as a register.

Apart from that it is hard to know which timestamp you want, but anyhow please read the server manual, or file an issue with that project...it does not seem to have anything to do with pymodbus,

closed 2024-04-16 by janiversen

shamyjun22, 2024-04-18:
Thanks for you reply. I have read the manual. It seems that Kepware doesn't support date type register addressing. I can now generate datetime, but the problem is it in string format. It is possible for pymodbus to handle string data type?

janiversen, 2024-04-18:
Pymodbus handles registers, what you do with the data is up to you.

shamyjun22, 2024-04-21:
> Pymodbus handles registers, what you do with the data is up to you.

Thanks. It is clear to me now.