pymodbus-dev/pymodbus #2173: TCP client frame mismatch under heavy load

martyy665, 2024-04-22:
### Versions

- Python: 3.11
- OS: Linux
- Pymodbus:  3.6.8

### Pymodbus Specific

- Client: tcp - async

### Description

After some periodic readouts from multiple slaves, wrong data are returned (data coming from another slave). It appears when the `tid` counter overflows 65535 and goes back to zero. Changing the line 507 in `pymodbus/transaction.py` below fixes the issue.

### Code and Logs

```python

     def getTransaction(self, tid):
499         """Return a transaction matching the referenced tid.
500
501         If the transaction does not exist, None is returned
502
503         :param tid: The transaction to retrieve
504
505         """
506         Log.debug("Getting transaction {}", tid)
507         if tid is None:
508             if self.transactions:
509                 return self.transactions.popitem()[1]
510             return None
511         return self.transactions.pop(tid, None)

```

janiversen, 2024-04-22:
Please submit a pull request.

It is an interesting observation, we do not have a test case for this.

martyy665, 2024-04-23:
Closed by approved PR.

janiversen, 2024-04-23:
It seems I slept when approving this, because it breaks a lot of tests. Problem is that tid == 0 is an allowed transaction,

Will submit a fix shortly,