pymodbus-dev/pymodbus #2182: incorrect register reading

Alfick, 2024-04-27:
Good afternoon, I am facing this problem when I write certain values [29339, 56070, 34472, 65077, 46577, 24805, 18085, 23349, 11601, 6050, 6656] to the device, then during reading I get [155, 6, 168, 53, 241, 221, 229, 165, 53, 81, 162, 0], here is the device log 01 10 ff b0 00 0b 16 72 9b db 06 86 a8 fe 35 b5
f1 60 e5 46 a5 5b 35 2d 51 17 a2 1a 00 08 78, and it turns out that while reading the register it only gives the value of the second byte(example: 729b-29339, 9b-155), what do I need to do to get [29339, 56070, 34472, 65077, 46577, 24805, 18085, 23349, 11601, 6050, 6656].

import asyncio
from pymodbus.client import ModbusSerialClient, AsyncModbusSerialClient

def test_1():
    client = ModbusSerialClient(port='COM6', baudrate=9600, stopbits=2, parity='N', timeout=0.5, retries=1)
    client.connect()
    res = client.read_holding_registers(1024, slave=1)
    print(res.registers)
    client.write_registers(65456, [29339, 56070, 34472, 65077, 46577, 24805, 18085, 23349, 11601, 6050, 6656], slave=1)
    res1 = client.read_holding_registers(65456, count=11, slave=1)
    print(res1.registers)
    res2 = client.read_holding_registers(65456, count=1, slave=1)
    print(res2.registers)
    client.close()

test_1()

output:
[113]
[155, 6, 168, 53, 241, 229, 165, 53, 81, 162, 0]
[155]

janiversen, 2024-04-27:
A register is 16bit and that is what pymodbus return, in order to to see what actually happens please run it with the pymodbus debug logging, then we can see what is received from the device and compare it to what the app receives.

Your code looks ok.

Alfick, 2024-04-28:
import asyncio
import logging
from pymodbus.client import ModbusSerialClient, AsyncModbusSerialClient

logging.basicConfig()
log = logging.getLogger()
log.setLevel(logging.DEBUG)

def test_1():
    client = ModbusSerialClient(port='COM6', baudrate=9600, stopbits=2, parity='N', timeout=0.5, retries=1)
    client.connect()
    # res = client.read_holding_registers(1024, slave=1)
    # print(res.registers)
    # client.write_registers(65456, [29339, 56070, 34472, 65077, 46577, 24805, 18085, 23349, 11601, 6050, 6656], slave=1)
    # res1 = client.read_holding_registers(65456, count=11, slave=1)
    # print(res1.registers)
    res2 = client.read_holding_registers(65456, count=11, slave=1)
    print(res2.registers)
    client.close()

test_1()

DEBUG:pymodbus.logging:Current transaction state - IDLE
DEBUG:pymodbus.logging:Running transaction 1
DEBUG:pymodbus.logging:SEND: 0x1 0x3 0xff 0xb0 0x0 0xb 0x35 0xfe
DEBUG:pymodbus.logging:Resetting frame - Current Frame in buffer - 
DEBUG:pymodbus.logging:New Transaction state "SENDING"
DEBUG:pymodbus.logging:Changing transaction state from "SENDING" to "WAITING FOR REPLY"
[155, 6, 168, 53, 241, 229, 165, 53, 81, 162, 0]
DEBUG:pymodbus.logging:Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
DEBUG:pymodbus.logging:RECV: 0x1 0x3 0x16 0x0 0x9b 0x0 0x6 0x0 0xa8 0x0 0x35 0x0 0xf1 0x0 0xe5 0x0 0xa5 0x0 0x35 0x0 0x51 0x0 0xa2 0x0 0x0 0xef 0x3b
DEBUG:pymodbus.logging:Processing: 0x1 0x3 0x16 0x0 0x9b 0x0 0x6 0x0 0xa8 0x0 0x35 0x0 0xf1 0x0 0xe5 0x0 0xa5 0x0 0x35 0x0 0x51 0x0 0xa2 0x0 0x0 0xef 0x3b
DEBUG:pymodbus.logging:Getting Frame - 0x3 0x16 0x0 0x9b 0x0 0x6 0x0 0xa8 0x0 0x35 0x0 0xf1 0x0 0xe5 0x0 0xa5 0x0 0x35 0x0 0x51 0x0 0xa2 0x0 0x0
DEBUG:pymodbus.logging:Factory Response[ReadHoldingRegistersResponse': 3]
DEBUG:pymodbus.logging:Frame advanced, resetting header!!
DEBUG:pymodbus.logging:Adding transaction 1
DEBUG:pymodbus.logging:Getting transaction 1
DEBUG:pymodbus.logging:Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"

Process finished with exit code 0

janiversen, 2024-04-28:
Please have a look at what pymodbus receives:

DEBUG:pymodbus.logging:RECV: 0x1 0x3 0x16 0x0 0x9b 0x0 0x6 0x0 0xa8 0x0 0x35 0x0 0xf1 0x0 0xe5 0x0 0xa5 0x0 0x35 0x0 0x51 0x0 0xa2 0x0 0x0 0xef 0x3b

0x1 = slave
0x3 = ReadHolding registers
0x16 = length in bytes of payload
And then the registers:
0x0 0x9b
0x0 0x6
0x0 0xa8
0x0 0x35
0x0 0xf1
0x0 0xe5
0x0 0xa5
0x0 0x35
0x0 0x51
0x0 0xa2
0x0 0x0

It seems your device have decided to always send 0x0 in the first byte, could it be that your addressing is off by 1.

closed 2024-06-03 by github-actions[bot]