pymodbus-dev/pymodbus #2188: consider changing the github default landing branch to "master" ?

chihpingkuo, 2024-05-08:
### Description

When I first visit this github project, I look into example folder, execute one example, and expect it to run without problem
however, I encounter an import error instead.
It took me some time to realize that the example I copied was from the default branch "dev", which currently conflict with the package I installed with "pip install pymodbus" (the master branch ?)

### Suggestion
consider changing the github default landing branch to "master" ?

janiversen, 2024-05-08:
If you miss a file, it's because you did not install the whole directory examples, please read the documentation! we provide a few examples simple_ that run as standalone the rest depend on helper.py

This has nothing to do with branches, "pip install pymodbus" do NOT install examples.

Changing to master would be a very dangerous default, because that would be the base for all pull requests, but they must be merged to "dev", "master" only contains releases.

janiversen, 2024-06-12:
We do not want to change the default to master, since it would mean that Pull Requests would be raised against master, which is wrong.

closed 2024-06-12 by janiversen