pymodbus-dev/pymodbus #2196: target machine actively refused connection

amarco, 2024-05-14:
I have a an odd on. I am polling a device once an hour, each hour I read/process 120 registers of length 2.
I open and close the connect for every parameter.
Everything work ducky except in mid stream I will get 
....snippet
024-05-14 08:57:15 | [INFO] | ..Processing parameter: L3kVA
2024-05-14 08:57:19 | [INFO] | Processing load: 5 of 11
2024-05-14 08:57:19 | [INFO] | ..Processing parameter: load
2024-05-14 08:57:21 | [INFO] | ..Processing parameter: isEnabled
2024-05-14 08:57:25 | [INFO] | ..Processing parameter: Vref
2024-05-14 08:57:27 | [INFO] | ..Processing parameter: totalAmps
2024-05-14 08:57:30 | [INFO] | ..Processing parameter: totalkWh
2024-05-14 08:57:33 | [INFO] | ..Processing parameter: totalkW
2024-05-14 08:57:36 | [INFO] | ..Processing parameter: L1Amps
2024-05-14 08:57:39 | [INFO] | ..Processing parameter: L2Amps
2024-05-14 08:57:42 | [INFO] | ..Processing parameter: L3Amps
2024-05-14 08:57:45 | [INFO] | ..Processing parameter: L1kVA
2024-05-14 08:57:48 | [INFO] | ..Processing parameter: L2kVA
2024-05-14 08:57:50 | [INFO] | ..Processing parameter: L3kVA
2024-05-14 08:57:54 | [INFO] | Processing load: 6 of 11
2024-05-14 08:57:54 | [INFO] | ..Processing parameter: load
2024-05-14 08:57:57 | [ERROR] | Connection to (10.10.4.167, 502) failed: [WinError 10061] No connection could be made because the target machine actively refused it

I am wondering if there is a more forcefully way to disconnect and ensure no connections happen or is there some other problem afoot.

My getdata method is below...

def GetData(regStart,numOfReg, returnType):
    
    client = ModbusTcpClient(host="redacted",port=502, framer=Framer.SOCKET, retries=3, timeout=30 )
    time.sleep(1)
    try:
        client.connect()

        if not client.connected:
            log.info(f"...Client error: {client.error_received()}")
            log.info("...Modbus TCP client could not connect; Exiting")
            sys.exit(1)
            
    except SocketError as e:
        log.info(f"...Modbus exception: {e} ; Exiting")
        sys.exit(1)
    
    try:
        regList = client.read_input_registers(address=regStart,count=numOfReg,slave=1)

        if not client.connected:
            log.info(f"...Client disconnected during read {client.error_received()}")
            sys.exit(1)
        
        if regList is None:
            log.info(f"...no register(s) returned; Exiting")
            sys.exit(1)

        client.close()

        match returnType.name:
             case ReturnType.INT.name:       
                 return BinaryPayloadDecoder.fromRegisters(regList.registers, Endian.BIG, wordorder=Endian.BIG).decode_32bit_int()
             case ReturnType.FLOAT.name:                 
                 return BinaryPayloadDecoder.fromRegisters(regList.registers, Endian.BIG, wordorder=Endian.BIG).decode_32bit_float()
             case ReturnType.STRING.name:        
                 return codecs.decode(BinaryPayloadDecoder.fromRegisters(regList.registers, Endian.BIG, wordorder=Endian.BIG)._payload,'utf-8').rstrip('\x00')                 
             case ReturnType.BOOL.name:        
                 return bool(BinaryPayloadDecoder.fromRegisters(regList.registers, Endian.BIG, wordorder=Endian.BIG).decode_32bit_int())     
             case _:
                 log.info(f"...unable to decode data; Exiting")
                 sys.exit(1)
        
    except ModbusIOException as e:
        log.info(f"...Modbus exception: {e} ; Exiting")
        sys.exit(1)

janiversen, 2024-05-14:
Please run it with the pymodbus debug call, so we can see what happens.

Your code looks ok, but it is very unusual to disconnect/connect with every call.

janiversen, 2024-05-14:
Your reglist wrong, in case of an error reglist is not none, but contains error info, please see our documentation

amarco, 2024-05-14:
Sorry artifact of previous version which used different library will fix when back in office.  Will do that and also generate debug.

RWDI - A Platinum Member of Canada's 50 Best Managed Companies
This communication is intended for the sole use of the party to whom it was addressed and may contain information that is privileged and/or confidential. Any other distribution, copying or disclosure is strictly prohibited. If you received this email in error, please notify us immediately by replying to this email and delete the message without retaining any hard or electronic copies of same. Outgoing emails are scanned for viruses, but no warranty is made to their absence in this email or attachments. If you require any information supplied by RWDI in a different format to facilitate accessibility, contact the sender of the email, email ***@***.*** or call +1.519.823.1311.
Please be aware that when you contact us with a business query we may collect and use your details for future communications.

amarco, 2024-05-14:
Interesting, when it passes the server it thinks it connected to is my local machine " Socket ('10.10.34.76', 48533)" and when it fails it is trying to talk to "Connection to (10.10.4.167, 502) failed" not sure what is happening here?

Marco

2024-05-14 13:51:30 | [DEBUG] | [16149, 4414]
2024-05-14 13:51:30 | [DEBUG] | handle: [b'?\x15', b'\x11>']
2024-05-14 13:51:30 | [INFO] | ..Processing parameter: totalkWh
2024-05-14 13:51:31 | [DEBUG] | Connection to Modbus server established. Socket ('10.10.34.76', 48533)
2024-05-14 13:51:31 | [DEBUG] | Current transaction state - IDLE
2024-05-14 13:51:31 | [DEBUG] | Running transaction 1
2024-05-14 13:51:31 | [DEBUG] | SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x6 0xbe 0x0 0x2
2024-05-14 13:51:31 | [DEBUG] | New Transaction state "SENDING"
2024-05-14 13:51:31 | [DEBUG] | Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2024-05-14 13:51:32 | [DEBUG] | Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2024-05-14 13:51:32 | [DEBUG] | RECV: 0x0 0x1 0x0 0x0 0x0 0x7 0x1 0x4 0x4 0x45 0x94 0xed 0x33
2024-05-14 13:51:32 | [DEBUG] | Processing: 0x0 0x1 0x0 0x0 0x0 0x7 0x1 0x4 0x4 0x45 0x94 0xed 0x33
2024-05-14 13:51:32 | [DEBUG] | Factory Response[ReadInputRegistersResponse': 4]
2024-05-14 13:51:32 | [DEBUG] | Adding transaction 1
2024-05-14 13:51:32 | [DEBUG] | Frame check, no more data!
2024-05-14 13:51:32 | [DEBUG] | Getting transaction 1
2024-05-14 13:51:32 | [DEBUG] | Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
2024-05-14 13:51:32 | [DEBUG] | [17812, 60723]
2024-05-14 13:51:32 | [DEBUG] | handle: [b'E\x94', b'\xed3']
2024-05-14 13:51:32 | [INFO] | ..Processing parameter: totalkW
2024-05-14 13:51:34 | [ERROR] | Connection to (10.10.4.167, 502) failed: [WinError 10061] No connection could be made because the target machine actively refused it

janiversen, 2024-05-14:
Are you sure you read that correctly?

the server socket is 502, but the client socket can be any number.

janiversen, 2024-05-14:
Apart from that, your device is refusing the connection, probably because you make too many connections/disconnect in a very short time (most devices do not like that)

You really should do
- open connection
- read all registers
- close connection

to test it without changing your code you can add a larger delay between each connect call.

amarco, 2024-05-14:
I had thought of that and in fact that's the way I was doing but using a different modbus python library.
When I switched to pymodbus I did the open/close for every get data cycle.
I reworked the code to do a single open, read all parameters for all loads and then close.
I left debug on, and will let run. I also add a short sleep of 0.3 sec between each  get_data.
Will see share what happens.

Marco

janiversen, 2024-05-14:
If you keep the connection open there are NO reason for a sleep between read calls.

amarco, 2024-05-14:
Understood, I removed the sleep. So far I got 2 clean sets of data, now I have removed the sleep, lets see how that goes during the evening 😊

amarco, 2024-05-15:
Well things went  good till they did not and then it never recovered. Logs at time of failure, once it happens the only way to recover as far as I have found thus far is to reboot the modbus device I am polling.
Any thoughts.  I keep thinking it exhausting tcp/ip resources but open/closing connection should of helped. I am new to modbus and find all this perplexing that it would work for hours and then burp out...
Any thoughts are appreciated..

2024-05-14 17:55:05 | [INFO] | Collecting Server Room Power Levels
2024-05-14 17:55:05 | [DEBUG] | Connection to Modbus server established. Socket ('10.10.30.28', 54241)
2024-05-14 17:55:05 | [DEBUG] | Current transaction state - IDLE
2024-05-14 17:55:05 | [DEBUG] | Running transaction 1
2024-05-14 17:55:05 | [DEBUG] | SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0x0 0x56 0x0 0x2
2024-05-14 17:55:05 | [DEBUG] | New Transaction state "SENDING"
2024-05-14 17:55:05 | [DEBUG] | Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2024-05-14 17:55:06 | [DEBUG] | Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2024-05-14 17:55:06 | [DEBUG] | RECV: 0x0 0x1 0x0 0x0 0x0 0x7 0x1 0x4 0x4 0x0 0x0 0x0 0xb
2024-05-14 17:55:06 | [DEBUG] | Processing: 0x0 0x1 0x0 0x0 0x0 0x7 0x1 0x4 0x4 0x0 0x0 0x0 0xb
2024-05-14 17:55:06 | [DEBUG] | Factory Response[ReadInputRegistersResponse': 4]
2024-05-14 17:55:06 | [DEBUG] | Adding transaction 1
2024-05-14 17:55:06 | [DEBUG] | Frame check, no more data!
2024-05-14 17:55:06 | [DEBUG] | Getting transaction 1
2024-05-14 17:55:06 | [DEBUG] | Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
2024-05-14 17:55:06 | [DEBUG] | [0, 11]
2024-05-14 17:55:06 | [DEBUG] | handle: [b'\x00\x00', b'\x00\x0b']
2024-05-14 17:55:06 | [INFO] | Found: 11 loads
2024-05-14 17:55:06 | [INFO] | Processing load: 1 of 11
2024-05-14 17:55:06 | [INFO] | ..Processing parameter: load
2024-05-14 17:55:06 | [DEBUG] | Current transaction state - TRANSACTION_COMPLETE
2024-05-14 17:55:06 | [DEBUG] | Running transaction 2
2024-05-14 17:55:06 | [DEBUG] | SEND: 0x0 0x2 0x0 0x0 0x0 0x6 0x1 0x4 0x1 0x2e 0x0 0x14
2024-05-14 17:55:06 | [DEBUG] | New Transaction state "SENDING"
2024-05-14 17:55:06 | [DEBUG] | Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2024-05-14 17:55:07 | [DEBUG] | Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2024-05-14 17:55:07 | [DEBUG] | RECV: 0x0 0x2 0x0 0x0 0x0 0x2b 0x1 0x4 0x28 0x54 0x58 0x20 0x36 0x30 0x30 0x2d 0x32 0x30 0x38 0x56 0x20 0x33 0xc2 0xb0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2024-05-14 17:55:07 | [DEBUG] | Processing: 0x0 0x2 0x0 0x0 0x0 0x2b 0x1 0x4 0x28 0x54 0x58 0x20 0x36 0x30 0x30 0x2d 0x32 0x30 0x38 0x56 0x20 0x33 0xc2 0xb0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0 0x0
2024-05-14 17:55:07 | [DEBUG] | Factory Response[ReadInputRegistersResponse': 4]
2024-05-14 17:55:07 | [DEBUG] | Adding transaction 2
2024-05-14 17:55:07 | [DEBUG] | Frame check, no more data!
2024-05-14 17:55:07 | [DEBUG] | Getting transaction 2
2024-05-14 17:55:07 | [DEBUG] | Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
2024-05-14 17:55:07 | [DEBUG] | [21592, 8246, 12336, 11570, 12344, 22048, 13250, 45056, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0]
2024-05-14 17:55:07 | [INFO] | ..Processing parameter: isEnabled
2024-05-14 17:55:07 | [DEBUG] | Current transaction state - TRANSACTION_COMPLETE
2024-05-14 17:55:07 | [DEBUG] | Running transaction 3
2024-05-14 17:55:07 | [DEBUG] | SEND: 0x0 0x3 0x0 0x0 0x0 0x6 0x1 0x4 0x1 0x42 0x0 0x2
2024-05-14 17:55:07 | [DEBUG] | New Transaction state "SENDING"
2024-05-14 17:55:07 | [DEBUG] | Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2024-05-14 17:55:11 | [DEBUG] | Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
2024-05-14 17:55:11 | [DEBUG] | RECV: 0x0 0x3 0x0 0x0 0x0 0x7 0x1 0x4 0x4 0x0 0x0 0x0 0x1
2024-05-14 17:55:11 | [DEBUG] | Processing: 0x0 0x3 0x0 0x0 0x0 0x7 0x1 0x4 0x4 0x0 0x0 0x0 0x1
2024-05-14 17:55:11 | [DEBUG] | Factory Response[ReadInputRegistersResponse': 4]
2024-05-14 17:55:11 | [DEBUG] | Adding transaction 3
2024-05-14 17:55:11 | [DEBUG] | Frame check, no more data!
2024-05-14 17:55:11 | [DEBUG] | Getting transaction 3
2024-05-14 17:55:11 | [DEBUG] | Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
2024-05-14 17:55:11 | [DEBUG] | [0, 1]
2024-05-14 17:55:11 | [DEBUG] | handle: [b'\x00\x00', b'\x00\x01']
2024-05-14 17:55:11 | [INFO] | ..Processing parameter: Vref
2024-05-14 17:55:11 | [DEBUG] | Current transaction state - TRANSACTION_COMPLETE
2024-05-14 17:55:11 | [DEBUG] | Running transaction 4
2024-05-14 17:55:11 | [DEBUG] | SEND: 0x0 0x4 0x0 0x0 0x0 0x6 0x1 0x4 0x1 0x5e 0x0 0x2
2024-05-14 17:55:11 | [DEBUG] | New Transaction state "SENDING"
2024-05-14 17:55:11 | [DEBUG] | Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2024-05-14 17:55:30 | [DEBUG] | Transaction failed. ([WinError 10054] An existing connection was forcibly closed by the remote host)
2024-05-14 17:55:30 | [DEBUG] | Processing:
2024-05-14 17:55:30 | [DEBUG] | Frame check, no more data!
2024-05-14 17:55:30 | [DEBUG] | Getting transaction 4
2024-05-14 17:55:30 | [DEBUG] | Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
2024-05-14 17:55:30 | [INFO] | ...Ooops Client disconnected after read
[... cut ...]
2024-05-14 22:55:25 | [INFO] | ...Modbus TCP client could not connect; Exiting
2024-05-14 23:55:05 | [INFO] | Collecting Server Room Power Levels
2024-05-14 23:55:26 | [ERROR] | Connection to (10.10.4.167, 502) failed: [WinError 10060] A connection attempt failed because the connected party did not properly respond after a period of time, or established connection failed because connected host has failed to respond
2024-05-14 23:55:26 | [INFO] | ...Client error
2024-05-14 23:55:26 | [INFO] | ...Modbus TCP client could not connect; Exiting

janiversen, 2024-05-15:
This does not seem to be a modbus problem, but a tcp/ip problem.

Communicating on an open socket, does NOT use extra tcp/ip resources, but every connect uses quite a lot, and some devices are very slow in releasing these resources when the socket is closed.

Typically resources are not released before after 30 seconds !!! so if you need to ensure you do not connect/disconnect within that interval. Some devices take even longer to release the resources used by the socket (macOS as an example takes up to 2 hours).

The normal way is to connect once, and then simply read/write....the retries parameter ensure the library automatically reconnects when necessary. The reason I suggested a middle way was because I understood you wanted to disconnect/connect.

My production system (connection to a solar inverter) have been "connected" for more than 2months, in which time the router have rebooted several timer, but reconnect happen within the library. I do use the async version, which is a bit more modern, but I use to use the sync version without ever bothering to connect (apart from the initial one, during setup).

amarco, 2024-05-15:
I am tending to agree as no code pattern has been able to solve the problem.
I have sent email to the maker of the meter I am using to see if perhaps a firmware upgrade exists or they can shed some insights into this.
Thankyou for your comments.

janiversen, 2024-05-15:
Closing as this is not a pymodbus bug.

closed 2024-06-12 by janiversen