pymodbus-dev/pymodbus #2198: FramerType module not available with pymodbus 3.6.8.

Joy-2612, 2024-05-16:
#!/usr/bin/env python3
"""Pymodbus synchronous client example.

An example of a single threaded synchronous client.

usage: simple_sync_client.py

All options must be adapted in the code
The corresponding server must be started before e.g. as:
    python3 server_sync.py
"""

# --------------------------------------------------------------------------- #
# import the various client implementations
# --------------------------------------------------------------------------- #
import pymodbus.client as ModbusClient
from pymodbus import (
    ExceptionResponse,
    FramerType,
    ModbusException,
    pymodbus_apply_logging_config,
)

def run_sync_simple_client(comm, host, port, framer=FramerType.SOCKET):
    """Run sync client."""
    # activate debugging
    pymodbus_apply_logging_config("DEBUG")

    print("get client")
    if comm == "tcp":
        client = ModbusClient.ModbusTcpClient(
            host,
            port=port,
            framer=framer,
            # timeout=10,
            # retries=3,
            # retry_on_empty=False,y
            # source_address=("localhost", 0),
        )
    elif comm == "udp":
        client = ModbusClient.ModbusUdpClient(
            host,
            port=port,
            framer=framer,
            # timeout=10,
            # retries=3,
            # retry_on_empty=False,
            # source_address=None,
        )
    elif comm == "serial":
        client = ModbusClient.ModbusSerialClient(
            port,
            framer=framer,
            # timeout=10,
            # retries=3,
            # retry_on_empty=False,
            # strict=True,
            baudrate=9600,
            bytesize=8,
            parity="N",
            stopbits=1,
            # handle_local_echo=False,
        )
    elif comm == "tls":
        client = ModbusClient.ModbusTlsClient(
            host,
            port=port,
            framer=FramerType.TLS,
            # timeout=10,
            # retries=3,
            # retry_on_empty=False,
            # sslctx=None,
            certfile="../examples/certificates/pymodbus.crt",
            keyfile="../examples/certificates/pymodbus.key",
            # password=None,
            server_hostname="localhost",
        )
    else:
        print(f"Unknown client {comm} selected")
        return

    print("connect to server")
    client.connect()

    print("get and verify data")
    try:
        rr = client.read_coils(1, 1, slave=1)
    except ModbusException as exc:
        print(f"Received ModbusException({exc}) from library")
        client.close()
        return
    if rr.isError():
        print(f"Received Modbus library error({rr})")
        client.close()
        return
    if isinstance(rr, ExceptionResponse):
        print(f"Received Modbus library exception ({rr})")
        # THIS IS NOT A PYTHON EXCEPTION, but a valid modbus message
        client.close()

    print("close connection")
    client.close()

if __name__ == "__main__":
    run_sync_simple_client("tcp", "127.0.0.1", "5020")
    
    
   The FramerType Module is not available for the pymodbus version 3.6.8.
   Instead we can use Framer module from the same version for implementing the same logic.

janiversen, 2024-05-16:
You copied the examples.tgz/zip from "latest" and not from v3.6.8, that is causing the problem.

"latest" correspond to v3.7.0dev which is the newest but unreleased code.

majki09, 2024-05-27:
I have also the same problem. Even if I get the example code from github or wiki, none of these works.

## simple_async_client.py:
```
pi@raspberry:~/pymodbus/examples $ python3.12 ./simple_async_client.py
Traceback (most recent call last):
  File "/home/pi/pymodbus/examples/./simple_async_client.py", line 15, in <module>
    from pymodbus import (
ImportError: cannot import name 'FramerType' from 'pymodbus' (/home/pi/.local/lib/python3.12/site-packages/pymodbus/__init__.py)
```

also this example also has some problem (helper.py is in the same folder)
## client_async.py:
```
pi@raspberry:~/pymodbus/examples $ python3 ./client_async.py -h
*** ERROR --> THIS EXAMPLE needs the example directory, please see
          https://pymodbus.readthedocs.io/en/latest/source/examples.html
          for more information.
```

Why do I see "latest" label on the 3.6.8?

janiversen, 2024-05-27:
You are mixing examples from dev with runtime from an earlier version, that will not work.

Ensure to download the examples from read the doc or github that matches your pymodbus version.

janiversen, 2024-06-12:
Closing as not being an issue.

closed 2024-06-12 by janiversen