pymodbus-dev/pymodbus #2218: pymodbus 3.6.8 does not work, while works with modbus-tk and pymodbus 2.5.3

rishabh1212, 2024-06-14:
### Versions

- Python: 3.12
- OS: Linux
- Pymodbus: 2.5.3 & 3.6.8
- Modbus Hardware (if used):
- Keyence PLC KV-8000 Modbus software

### Pymodbus Specific

- Server: TCP
- Client: TCP

### Description

### Code and Logs

#### Working for pymodbus 2.5.3

```python
import time
import logging
from pymodbus.client.sync import ModbusTcpClient
from pymodbus.constants import Defaults

# Enable logging
logging.basicConfig()
logger = logging.getLogger()
logger.setLevel(logging.DEBUG)

# Create Modbus client
client = ModbusTcpClient('192.168.0.11', port=502, timeout=5.0)

while True:
    try:
        time.sleep(2)
        response = client.read_input_registers(4001, 1, unit=1)
        
        if not response.isError():
            print("Register value:", response.registers[0])
        else:
            logger.error(f"Error reading input register: {response}")
        
        time.sleep(0.1)
        
        # To write to a register, uncomment and use the following line:
        # client.write_register(4100, response.registers[0], unit=1)
        
    except Exception as e:
        logger.error(str(e))
        client.close()
        break
```

Pymodbus 2.5.3 Working log
```
DEBUG:pymodbus.client.sync:Connection to Modbus server established. Socket ('192.168.0.55', 58178)
DEBUG:pymodbus.transaction:Current transaction state - IDLE
DEBUG:pymodbus.transaction:Running transaction 1
DEBUG:pymodbus.transaction:SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x1 0x4 0xf 0xa1 0x0 0x1
DEBUG:pymodbus.client.sync:New Transaction state 'SENDING'
DEBUG:pymodbus.transaction:Changing transaction state from 'SENDING' to 'WAITING FOR REPLY'
DEBUG:pymodbus.transaction:Changing transaction state from 'WAITING FOR REPLY' to 'PROCESSING REPLY'
DEBUG:pymodbus.transaction:RECV: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x0 0x2d
DEBUG:pymodbus.framer.socket_framer:Processing: 0x0 0x1 0x0 0x0 0x0 0x5 0x1 0x4 0x2 0x0 0x2d
DEBUG:pymodbus.factory:Factory Response[ReadInputRegistersResponse: 4]
DEBUG:pymodbus.transaction:Adding transaction 1
DEBUG:pymodbus.transaction:Getting transaction 1
DEBUG:pymodbus.transaction:Changing transaction state from 'PROCESSING REPLY' to 'TRANSACTION_COMPLETE'
Register value: 45
[... 153 lines cut ...]
DEBUG:pymodbus.logging:Processing: 
DEBUG:pymodbus.logging:Frame check, no more data!
DEBUG:pymodbus.logging:Getting transaction 4
DEBUG:pymodbus.logging:Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
ERROR:root:Error reading input register: Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response
DEBUG:pymodbus.logging:Connection to Modbus server established. Socket ('192.168.0.55', 58280)
DEBUG:pymodbus.logging:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.logging:Running transaction 5
DEBUG:pymodbus.logging:SEND: 0x0 0x5 0x0 0x0 0x0 0x6 0x0 0x4 0xf 0xa1 0x0 0x1
DEBUG:pymodbus.logging:New Transaction state "SENDING"
DEBUG:pymodbus.logging:Changing transaction state from "SENDING" to "WAITING FOR REPLY"
DEBUG:pymodbus.logging:No response received to unbounded read !!!!
DEBUG:pymodbus.logging:Changing transaction state from "WAITING FOR REPLY" to "PROCESSING REPLY"
DEBUG:pymodbus.logging:RECV: 
DEBUG:pymodbus.logging:Processing: 
DEBUG:pymodbus.logging:Frame check, no more data!
DEBUG:pymodbus.logging:Getting transaction 5
DEBUG:pymodbus.logging:Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
ERROR:root:Error reading input register: Modbus Error: [Input/Output] No Response received from the remote slave/Unable to decode response
DEBUG:pymodbus.logging:Connection to Modbus server established. Socket ('192.168.0.55', 58281)
DEBUG:pymodbus.logging:Current transaction state - TRANSACTION_COMPLETE
DEBUG:pymodbus.logging:Running transaction 6
DEBUG:pymodbus.logging:SEND: 0x0 0x6 0x0 0x0 0x0 0x6 0x0 0x4 0xf 0xa1 0x0 0x1
DEBUG:pymodbus.logging:New Transaction state "SENDING"
```

janiversen, 2024-06-14:
Clearly it does not work, unit= no longer exist, did you try to read the documentation?

closed 2024-06-14 by janiversen

janiversen, 2024-06-14:
we have parameter documentation in the code, and a full documentation on readthedocs, so I do not really see the need for a warning......there are many other cases where people should read the documentation.

janiversen, 2024-06-14:
Ohhh forgot we have nearly replaced all kwargs, with named parameters among others yours.

rishabh1212, 2024-06-14:
Thanks!
I will keep a note of these from next time.