pymodbus-dev/pymodbus #2221: Example server_async.py: AttributeError: module 'helper' has no attribute 'get_commandline'

peterz1972, 2024-06-21:
- Python: 3.11.2
- OS: Raspbian GNU/Linux 12
- Pymodbus: 3.6.8

The example "server_async.py" imports "helper". So I installed helper 2.5.0 via pip.
In the example the function helper.get_commandline(...) gets called.
Helper does not have this function. I get this error:

AttributeError: module 'helper' has no attribute 'get_commandline'

I also searched in the older helper versions for it.
Did I install the wrong "helper"?
```

janiversen, 2024-06-22:
I do not know what helper 2.5.0 is, it is not something we use.

Please read the documentation, to use the advanced examples you need to copy the examples directory.

janiversen, 2024-06-22:
And just to be clear did you read the error message we provide ????

```
    print("*** ERROR --> THIS EXAMPLE needs the example directory, please see \n\
          https://pymodbus.readthedocs.io/en/latest/source/examples.html\n\
          for more information.")
```

closed 2024-06-22 by janiversen

peterz1972, 2024-06-22:
Oh! Thank you! Saved me maybe a lot of time - semi professionals must be
quite annoying sometimes - sorry ... :-/