pymodbus-dev/pymodbus #2229: Not working after upgrade tp 3.6.9

f41ardu, 2024-07-13:
### Versions

- Python: 3.10
- OS: Debian 6.7.9-2 (2024-03-13) x86_64 GNU/Linux
- Pymodbus: 3.6.9
- Modbus Hardware (if used): SUN2000-6KTL-M1 Softwareversion V100R001C00SPC162 

### Pymodbus Specific

- Client: tcp - sync (see my initial code below) 

### Description

Latest stable version: 3.6.9
Summary: A fully featured modbus protocol stack in python
Author: Galen Collins, Jan Iversen
Homepage: Could not find the package info from PyPI. Error code: wrong # args: should be ".!localcpythonpipdialog3.!frame.!panedwindow.!frame2.!textframe.!enhancedtext_orig insert index chars ?tagList chars tagList ...?"

Looks like that the package is broken (see above) 

I just start my implementation, But it was unstable, too.  But  after upgrade to latest release it didn't work any longer and I get no respon
se from tested register I tried before. 

### Code and Logs

```python
from pymodbus.client import ModbusTcpClient
from pymodbus.exceptions import ModbusIOException
import pymodbus

from time import sleep

pymodbus.pymodbus_apply_logging_config("DEBUG")

# Konfigurationsparameter
ip_adresse = '192.168.179.2'  # IP-Adresse des Wechselrichters
port = 502 # Modbus TCP Standardport
slave_id = 1  # Die Modbus-Adresse des Wechselrichters, oft 1

# Modbus-Client initialisieren
client = ModbusTcpClient(ip_adresse, port=port, timeout=.5)

try:
    # Verbindung zum Wechselrichter herstellen
    if client.connect():
        print(f"Verbindung zum Wechselrichter unter {ip_adresse}:{port} erfolgreich.")
        register_address = 32078  # 37758 #37022 #37118  # Das Register, das abgefragt werden soll
        register_count = 2  # Anzahl der Register, die abgefragt werden sollen
        sleep(2)
        response = client.read_holding_registers(register_address, register_count,  unit=slave_id)
        
        if not isinstance(response, ModbusIOException):
            # Daten aus der Antwort extrahieren
            register_value = response.registers
            print(f"Registerwert (Adresse {register_address}): {register_value}")
        else:
            print(f"Fehler beim Abfragen des Registers {register_address}.")
        
        # Verbindung trennen
        client.close()
    else:
        print(f"Verbindung zum Wechselrichter unter {ip_adresse}:{port} fehlgeschlagen.")
        client.close()
except Exception as e:
    print(f"Ein Fehler ist aufgetreten: {e}")
´´´
´´´
2024-07-13 12:07:04,560 DEBUG logging:103 Connection to Modbus server established. Socket ('192.168.179.6', 38310)
Verbindung zum Wechselrichter unter 192.168.179.2:502 erfolgreich.
2024-07-13 12:07:06,562 DEBUG logging:103 Current transaction state - IDLE
2024-07-13 12:07:06,563 DEBUG logging:103 Running transaction 1
2024-07-13 12:07:06,563 DEBUG logging:103 SEND: 0x0 0x1 0x0 0x0 0x0 0x6 0x0 0x3 0x7d 0x4e 0x0 0x2
2024-07-13 12:07:06,564 DEBUG logging:103 New Transaction state "SENDING"
2024-07-13 12:07:06,564 DEBUG logging:103 Changing transaction state from "SENDING" to "WAITING FOR REPLY"
2024-07-13 12:07:07,066 DEBUG logging:103 Transaction failed. (Modbus Error: [Invalid Message] No response received, expected at least 8 bytes (0 received)) 
2024-07-13 12:07:07,066 DEBUG logging:103 Processing: 
2024-07-13 12:07:07,067 DEBUG logging:103 Very short frame (NO MBAP):  wait for more data
2024-07-13 12:07:07,067 DEBUG logging:103 Getting transaction 1
2024-07-13 12:07:07,067 DEBUG logging:103 Changing transaction state from "PROCESSING REPLY" to "TRANSACTION_COMPLETE"
Fehler beim Abfragen des Registers 32078.
´´´

janiversen, 2024-07-13:
Please read the documentation you are using unknown/wrong parameters.

Please also be careful with using positional parameters for parameters declared as "by name", that is bound to cause a problem sooner or later.

f41ardu, 2024-07-13:
You mean timeout=25 and unit=1 ?  
 
Good point, but any idea where I can find examples.  

Thanks in advance

janiversen, 2024-07-13:
Yes examples are part of our documentation.

f41ardu, 2024-07-17:
It works with the following configuration from the contributed examples [solar.py](https://github.com/pymodbus-dev/pymodbus/blob/dev/examples/contrib/solar.py)

```
client = ModbusTcpClient(
        "192.168.179.2",
        port=502,
        # Common optional parameters:
        framer=ModbusSocketFramer,
        timeout=10,
        retries=3,
        retry_on_empty=True,   
    )
```
I'm not quite sure if it is necessary to open and close the connection for every read out of a set of registers. 

My Inverter (Huawai)
Model: SUN2000-6KTL-M1 
Softwareversion: V100R001C00SPC162 
Connected via WiFi Dongle
Model: SDongleA-05 
Softwareversion: V200R022C10SPC114

Connection via NAT from my local LAN to GuestLan (Fritzbox)

closed 2024-07-17 by f41ardu

janiversen, 2024-07-17:
It is not, actually it is a bad idea.

f41ardu, 2024-07-17:
Hey, what is the bad idea. Access via NAT. I wouldn't say so. Or open and close the connection? 
Data Transfer to fusion portal is still up and running.

janiversen, 2024-07-18:
Open/close with each call, most devices do not like that, and reject connection after a while.

But if it's works for then fine, but you did ask.