pymodbus-dev/pymodbus #2231: examples not working

zockerforever, 2024-07-14:
### Versions

- Python: 3.11.9
- OS: debian
- Pymodbus: newest
- Modbus Hardware (if used): usb-modbus adapter

### Pymodbus Specific

- Server: rtu async
- Client: rtu sync and rtu async

### Description

i started the example async server and tried the example clients. sync and async.
i expected that to work as the example code was cloned (git clone....)
i was trying to watch what happens using the debug log level, output attached below.
With the sinc client not much happens, with async clients alot of frame check failed/ frame not ready problems

### Code and Logs

```python
# code and logs here.
server logs:
python3 server_async.py -c serial -f rtu -p /dev/ttyUSB0 --baudrate 19200 -l debug 
2024-07-14 22:09:32,192 DEBUG logging:103 Awaiting connections server_listener
2024-07-14 22:09:32,193 INFO  logging:97 Server listening.
2024-07-14 22:09:32,193 DEBUG logging:103 Connected to server
2024-07-14 22:09:49,567 DEBUG logging:103 recv: 0x1 0x40 0x20 0x0 0x40 old_data:  addr=None
2024-07-14 22:09:49,568 DEBUG logging:103 Handling data: 0x1 0x40 0x20 0x0 0x40
2024-07-14 22:09:49,568 DEBUG logging:103 Processing: 0x1 0x40 0x20 0x0 0x40
2024-07-14 22:12:10,680 DEBUG logging:103 recv: 0x1 0x1 0x0 0x2 0x1 0xfc 0x0 old_data:  addr=None
2024-07-14 22:12:10,680 DEBUG logging:103 Handling data: 0x1 0x1 0x0 0x2 0x1 0xfc 0x0
2024-07-14 22:12:10,680 DEBUG logging:103 Processing: 0x20 0x0 0x40 0x1 0x1 0x0 0x2 0x1 0xfc 0x0
2024-07-14 22:12:10,680 DEBUG logging:103 Frame check failed, ignoring!!
2024-07-14 22:12:10,681 DEBUG logging:103 Resetting frame - Current Frame in buffer - 0x40 0x1 0x1 0x0 0x2 0x1 0xfc 0x0
2024-07-14 22:12:10,681 DEBUG logging:103 Frame - not ready
2024-07-14 22:12:20,688 DEBUG logging:103 recv: 0x1 0x1 0x20 0x0 0x1 0xfc 0x0 old_data:  addr=None
2024-07-14 22:12:20,689 DEBUG logging:103 Handling data: 0x1 0x1 0x20 0x0 0x1 0xfc 0x0
2024-07-14 22:12:20,689 DEBUG logging:103 Processing: 0x1 0x1 0x0 0x2 0x1 0xfc 0x0 0x1 0x1 0x20 0x0 0x1 0xfc 0x0
2024-07-14 22:12:20,689 DEBUG logging:103 Frame check failed, ignoring!!
2024-07-14 22:12:20,689 DEBUG logging:103 Resetting frame - Current Frame in buffer - 0x1 0x1 0x0 0x2 0x1 0xfc 0x0 0x1 0x1 0x20 0x0 0x1 0xfc 0x0
2024-07-14 22:12:20,689 DEBUG logging:103 Frame check failed, ignoring!!
2024-07-14 22:12:20,689 DEBUG logging:103 Resetting frame - Current Frame in buffer - 0x0 0x2 0x1 0xfc 0x0 0x1 0x1 0x20 0x0 0x1 0xfc 0x0
2024-07-14 22:12:20,689 DEBUG logging:103 Frame check failed, ignoring!!
2024-07-14 22:12:20,689 DEBUG logging:103 Resetting frame - Current Frame in buffer - 0x2 0x1 0xfc 0x0 0x1 0x1 0x20 0x0 0x1 0xfc 0x0
2024-07-14 22:12:20,689 DEBUG logging:103 Frame check failed, ignoring!!
2024-07-14 22:12:20,689 DEBUG logging:103 Resetting frame - Current Frame in buffer - 0x0 0x1 0x1 0x20 0x0 0x1 0xfc 0x0
2024-07-14 22:12:20,689 DEBUG logging:103 Frame - not ready
2024-07-14 22:12:30,698 DEBUG logging:103 recv: 0x1 0x1 0x0 0x2 0x1 0xfc 0x0 old_data:  addr=None
2024-07-14 22:12:30,699 DEBUG logging:103 Handling data: 0x1 0x1 0x0 0x2 0x1 0xfc 0x0
2024-07-14 22:12:30,699 DEBUG logging:103 Processing: 0x1 0x1 0x20 0x0 0x1 0xfc 0x0 0x1 0x1 0x0 0x2 0x1 0xfc 0x0
2024-07-14 22:12:30,699 DEBUG logging:103 Frame check failed, ignoring!!
2024-07-14 22:12:30,699 DEBUG logging:103 Resetting frame - Current Frame in buffer - 0x1 0x1 0x20 0x0 0x1 0xfc 0x0 0x1 0x1 0x0 0x2 0x1 0xfc 0x0
2024-07-14 22:12:30,700 DEBUG logging:103 Frame check failed, ignoring!!
2024-07-14 22:12:30,700 DEBUG logging:103 Resetting frame - Current Frame in buffer - 0x0 0x1 0xfc 0x0 0x1 0x1 0x0 0x2 0x1 0xfc 0x0
2024-07-14 22:12:30,700 DEBUG logging:103 Frame check failed, ignoring!!
2024-07-14 22:12:30,700 DEBUG logging:103 Resetting frame - Current Frame in buffer - 0x0 0x1 0x1 0x0 0x2 0x1 0xfc 0x0
2024-07-14 22:12:30,700 DEBUG logging:103 Frame - not ready
2024-07-14 22:12:40,708 DEBUG logging:103 recv: 0x40 0x40 0x2 0x1 0xfc old_data:  addr=None
2024-07-14 22:12:40,709 DEBUG logging:103 Handling data: 0x40 0x40 0x2 0x1 0xfc
2024-07-14 22:12:40,709 DEBUG logging:103 Processing: 0x1 0x1 0x0 0x2 0x1 0xfc 0x0 0x40 0x40 0x2 0x1 0xfc
2024-07-14 22:12:40,709 DEBUG logging:103 Frame check failed, ignoring!!
2024-07-14 22:12:40,709 DEBUG logging:103 Resetting frame - Current Frame in buffer - 0x1 0x1 0x0 0x2 0x1 0xfc 0x0 0x40 0x40 0x2 0x1 0xfc
2024-07-14 22:12:40,709 DEBUG logging:103 Frame check failed, ignoring!!
2024-07-14 22:12:40,709 DEBUG logging:103 Resetting frame - Current Frame in buffer - 0x0 0x2 0x1 0xfc 0x0 0x40 0x40 0x2 0x1 0xfc
2024-07-14 22:12:40,709 DEBUG logging:103 Frame check failed, ignoring!!
2024-07-14 22:12:40,710 DEBUG logging:103 Resetting frame - Current Frame in buffer - 0x2 0x1 0xfc 0x0 0x40 0x40 0x2 0x1 0xfc
2024-07-14 22:12:40,710 DEBUG logging:103 Frame - not ready

client sync:
python3 client_sync.py -c serial -f rtu -p /dev/ttyUSB1 --baudrate 19200                             
Traceback (most recent call last):
  File "/home/henri/code/modbusquick/pymodbus/examples/client_sync.py", line 155, in <module>
    main()
  File "/home/henri/code/modbusquick/pymodbus/examples/client_sync.py", line 151, in main
    run_sync_client(testclient, modbus_calls=run_a_few_calls)
  File "/home/henri/code/modbusquick/pymodbus/examples/client_sync.py", line 130, in run_sync_client
    modbus_calls(client)
  File "/home/henri/code/modbusquick/pymodbus/examples/client_sync.py", line 139, in run_a_few_calls
    assert len(rr.bits) == 8
               ^^^^^^^
AttributeError: 'ModbusIOException' object has no attribute 'bits'
                                                                    
client async produces no cli output itself

# please use the following to format logs when posting them here
import pymodbus

pymodbus.pymodbus_apply_logging_config("DEBUG")
```

janiversen, 2024-07-14:
The examples are working, seems a lot like you have a cabling problem, did you e.g. remember the minimum length and pull up resistors.

janiversen, 2024-07-14:
Please have a look at what the client sends and what the server receives....that should be identical and is not controlled by pymodbus

closed 2024-07-18 by janiversen