pymodbus-dev/pymodbus #2234: Add issue tracker / source code repository to the doc

sciunto, 2024-07-18:
Hi,

It would be useful for users to have a link to the code repository and the issue tracker in the documentation.

Thanks

janiversen, 2024-07-18:
We do have both:

https://pymodbus.readthedocs.io/en/latest/source/readme.html

we point to our github repository, what do you want more ?

closed 2024-07-18 by janiversen

sciunto, 2024-07-18:
In the menu to be visible. 

I searched for tracker in the doc and could not find anything. See https://pymodbus.readthedocs.io/en/latest/search.html?q=tracker&check_keywords=yes&area=default
Same for bug: https://pymodbus.readthedocs.io/en/latest/search.html?q=bug&check_keywords=yes&area=default#

janiversen, 2024-07-18:
Pull requests are welcome.

janiversen, 2024-07-18:
"tracker" and "bug" are  not the names most commonly used in open source. Since GitHub uses "issues" so the vast majority of open source follow that.